nRF24/RF24Mesh #103: Oled- and RF24Mesh

mrred2k, 2016-11-24:
Hi,
I am having a problem with my project using rf24 Mesh.
It seams that there is a compability issue regarding the oled libary for the esp8266 and the rf24mesh.
I am using this Libary for the i2C oled [https://github.com/squix78/esp8266-oled-ssd1306](url)
on the nodemcu v1.0. with platformio.
The esp8266 is just stoping at mesh.update call in the loop. No resetting of some sort. Just stops.
My connections are not the problem, the examples work great in the same setup.

Has anyone have simular issues? Or can sombody point me to an other working examples using an i2c oled and the rf24 mesh?

I did not add an example because this very hard to repeat with the i2c oled and everything. Compiling is fine. I have some compiler warnings but they are the same in the examples.

Avamander, 2016-11-25:
Does it stop at mesh.update() in the examples too?

mrred2k, 2016-11-25:
No the examples work wonderfully ( network and mesh) with the same hardware. That is my confusion and the reason why I failed to debug it.

Avamander, 2016-11-25:
Open the examples, make sure they work and then ONLY include the OLED
library, if it breaks, it's the OLED library messing something up and you
need to find an alternative.

mrred2k, 2016-11-28:
Had the chance to test it. Only including the ssd1306 libary in the rfmesh master example is resulting in constant restart when calling "display.clear()".

I gave up and redid my display in the adafruit_ssd1306 libary, where you can not change the i2c pins. But I got i running immidiately with the mesh working and the display and the BMP280 playing nicely.

closed 2017-01-03 by Avamander