nRF24/RF24Mesh #104: Node assignment problem in Mesh topology

chriskelman, 2016-11-28:
I have a nice nRF24 mesh system running with master on RPi and 6+ sensors running on Arduinos.
The mesh has been running mostly reliably for some months, but lately I find that the Node assignment has gone astray - all first level nodes are numbered 04, and all second level nodes are 044, so I now cannot tell how nodes are being relayed. (I have made sure that all nodes have unique nodeIDs assigned in the software)
In addition, nodes seem to have become unpredictable regarding connection - sometimes they connect instantly on restart, sometimes not at all. The next day, all will be ok.
I have tried deleting dhcplist.txt on the Pi (which is the only node running mesh.DHCP() ) and tried restarting all components of the system, but it makes no difference. My understanding is  that nothing is saved in EEPROM on the Arduino at shutdown, so I cannot explain this.
Any suggestions appreciated.

TMRh20, 2016-11-28:
Nodes typically should be checking and/or renewing their address regularly. I typically have my nodes do a ``` mesh.checkConnection() ``` every 30 seconds or so.

With this in mind, if you delete the correct dhcplist.txt file, from wherever the sketch is run, and leave the master/gateway offline for about 1 minute, all the nodes should pick up new addresses.

Also, see https://github.com/TMRh20/RF24/issues/307 for a simple example of how to manage channel hopping in case you are getting some interference when nodes are having trouble.

chriskelman, 2016-11-29:
OK thanks, tried all these suggestions, still stuck on node 04 and 044... I have checked for interference too and it is not significant at the upper channels away from WiFi.
There are two things I am still unclear about:
1) Use of DHCP() - I presume this should just be included in the master node script? (I have also just  noticed that the dhcp.txt file is no longer being created).
2) the difference between mesh.checkConnection() and mesh.update().
I have only been using mesh.checkConnnection() if a mesh.write() fails.
One of my node applications is time critical so I can't afford for this Arduino to be tied up seeking the mesh (I have set conditions when it is safe to do this), the other nodes can do this anytime.

TMRh20, 2016-12-01:
1. Yes. Either you have file creation/permissions problem or looking in the wrong place
2. CheckConnection() sends a ping, update() keeps the mesh update & handles routing

Might want to use a static RF24Network node for time-sensitive nodes. Will eventually make this a feature of RF24Mesh.

chriskelman, 2016-12-05:
Thanks for the suggestions, I have checked and found a couple of things but still not a solution. 
I am mystified as to what is going on with node addressing - it seems that the master station only allocates ID 04 but is prepared to do this for multiple senders. 
The problem I have with this is that I can't track what is going on, and the system is capricious about allowing new nodes to join.

I have searched and ensured there are no other instances of dhcplist.txt - in fact the program no longer seems to generate it. 
As for permissions, I did have an issue at reboot by starting the master program from root, but have eliminated that now.

I have just run the Master_Node program on the Pi with RF24Mesh_Example running on an Arduino.
I have mesh.setNodeID(0) on the RPi  and #define nodeID 1 on the Arduino.
All systems were off for 30 mins before I re-ran the programs - the output is below - if anyone has the time to make any suggestions I would appreciate it!

Arduino:
Connecting to the mesh...
Sender Node: 1
22 MSH: Poll 
28 MSH: Poll > -64dbm 
82 MSH: Poll OK 
82 MSH: Got poll from level 0 count 1
84 MSH: Req addr from 0
107 Set address:    4
Send OK: 1000....

On the RPi:
^Cpi@raspberrypi:~/kelman/Master_Node/Master_Node/bin/Debug $ sudo ./Master_Node 
start
================ SPI Configuration ================
CSN Pin  	 = CE0 (PI Hardware Driven)
CE Pin  	 = Custom GPIO22
Clock Speed	 = 8 Mhz
================ NRF Configuration ================
STATUS		 = 0x0e RX_DR=0 TX_DS=0 MAX_RT=0 RX_P_NO=7 TX_FULL=0
RX_ADDR_P0-1	 = 0xccccccccc3 0xcccccccc3c
RX_ADDR_P2-5	 = 0x33 0xce 0x3e 0xe3
TX_ADDR		 = 0xcccccc3ecc
RX_PW_P0-6	 = 0x20 0x20 0x20 0x20 0x20 0x20
EN_AA		 = 0x3e
EN_RXADDR	 = 0x3f
RF_CH		 = 0x61
RF_SETUP	 = 0x07
CONFIG		 = 0x0f
DYNPD/FEATURE	 = 0x3f 0x04
Data Rate	 = 1MBPS
Model		 = nRF24L01+
CRC Length	 = 16 bits
PA Power	 = PA_MAX
Rcv 1000 from 04
Rcv 2000 from 04....

closed 2019-05-10 by TMRh20