nRF24/RF24Mesh #105: RF24Mesh master-base node communication delay

somnathbm, 2016-12-06:
I'm trying to build a mesh network using 4 nRF24L01 modules directly connected with Arduino Uno where 1 module is master node and the rest are base nodes.

The workflow is exactly as follows

I'm sending a command in the Arduino IDE serial monitor. The master node receives this command and send it to other base nodes. There's no problem receiving the command on the base node side (from master node) as well as sending the command on the master node side (to base nodes). Upon receiving the command from the master node, base nodes successfully able to read the command and send the response. No problem at all.

But when the master node receives the response status from the base node(s), there's the problem. Master node is not receiving the data immediately. I have to send the command from the Arduino serial monitor repeatedly and on the 10th time, the master node is able to read the response status from the base node(s). Exactly on the 10th times every time.

Here's the link of the Arduino Stackexchange question thread.

http://arduino.stackexchange.com/questions/31964/rf24mesh-master-base-node-communication-delay

TMRh20, 2016-12-06:
RF24NetworkHeader header(mesh.addrList[j].address, OCT);
This line on master might be the problem.
Try replacing OCT with a 1 so master sends unacknowledged payload. Always a good idea if nodes are responding to requests, since ack is not necessary. (Node response is effectively an ack)

somnathbm, 2016-12-06:
@TMRh20 - I have changed the OCT to 1, as you just suggested. But the problem is still there.

TMRh20, 2016-12-06:
Ok I'll prob have to test to see. It might take me some time.

somnathbm, 2016-12-06:
Yeah that's fine.

I'm sending the command 100% or 0% in the Arduino serial monitor. If I send 100%, base node sends back response 1 and 0 if I send 0%. There's no problem in the base node(s). But master node always getting 1. I don't know where this 1 comes from. Master should get the value that the base is sending i.e. 1 or 0 respectively.

P.S. : This is happening from the beginning, even before the change from OCT to 1.

janLo, 2016-12-09:
I have a similar issue. I use the examples RF24Mesh_Example_Master_To_Nodes.ino and RF24Mesh_Example.ino for debugging and on the master after every sent packet the first N packets cannot be received. I spend two nights debugging and finally got a working solution without exact knowledge what I'm doing. My observation was, that the master was always broken after he sent a packet. So I concentrated myself on the switch between sending and receiving.

I got the best result (error almost completely gone) by adding a `delay(1)` before radio.startListen() at https://github.com/TMRh20/RF24Network/blob/master/RF24Network.cpp#L963.

I'm not exactly sure, what this changes, I guess, its some timing issue on the RF24. I'm also aware, that a delay is not a sufficient solution but maybe its a direction to debug into. My modules are bzw these: https://www.amazon.de/gp/product/B00XSYION0/ref=oh_aui_search_detailpage?ie=UTF8&psc=1

janLo, 2016-12-09:
PS: I use two Arduino Nano boards. @somnathbm , can you test, if this delay fixes your issue as well?

Avamander, 2016-12-09:
FYI I had problems like these with some fake modules. I gave up and bought
new ones.

ricmail85, 2016-12-31:
I have the same problem. The nodes can send messages to the master without any issue but the master is not able to write to the nodes any message. unfortunately the delay(1) suggested does not work for my configuration...
Any suggestion?

Avamander, 2016-12-31:
Does it hang? Do the messages get sent? More information would be
incredibly more helpful.

ricmail85, 2016-12-31:
The message is correctly received by the node, but on the master the result of mesh write is false. It doesn't hang.

Avamander, 2016-12-31:
You can implement custom check to verify that the message is received by
replying with something. It would take quite a bit of code more to do that
transparently and for most that is not an issue.

ricmail85, 2016-12-31:
Yes it is possible. 
But it's a strange behavior since in previous releases this issue was not present at all...

ricmail85, 2017-01-01:
Ok I found that it seems to be related to the line ok = radio.writeFast(frame_buffer, frame_size,0) in RF24Network.cpp. By using the previous version ok = radio.writeFast(frame_buffer, frame_size,multicast) there are much more messages that are correctly ack by the master. Instead by using ok = radio.writeFast(frame_buffer, frame_size,!multicast), suggested in another reply issue, the message are delivered but mesh.write results false to the master.

mamama1, 2017-01-15:
not being able to use 250kbps usually means counterfeits or not the + version of the NRF24L01.

ricmail85, 2017-01-15:
With the same modules I was able to send messages correctly even @ 
250kbps...
But something changed with latest releases of RF24, RF24Network and 
RF24Mesh I think...

Avamander, 2017-01-15:
Clones indeed have trouble with ACK, that might be why it isn't that stable.

closed 2020-08-16 by TMRh20