nRF24/RF24Mesh #107: Save and load DHCP

Pikokosan, 2016-12-17:
so i was looking at the code and thought. "wouldn't it be cool to save and load dhcp on a esp8266 using SPIFF" i've just started looking into it but figured i'd put the bug in someone's ear on here to see if they wanted to implement it.

Avamander, 2020-09-19:
Added this to the [Hacktoberfest](https://hacktoberfest.digitalocean.com/) list of potential contributions. 

Just to be clear what is expected from the feature:

* It should be implemented in a way that it can be enabled (only) on the ESP32 and ESP8266 using an `#define`. Support for other platforms that have writable flash is not in the scope.

* The table should be written periodically, the period should have a sane default, but it should be possible to change it.

* The DHCP table should only be written when it has been changed since the last time. Just reaching the write interval shouldn't trigger a pointless write (that most likely blocks for a while).

* The table should be read from flash during startup or when a function is called.

If someone else has any other ideas, suggestions, do comment them below and I'll maybe update the list above.

2bndy5, 2021-07-10:
> The table should be written periodically, the period should have a sane default, but it should be possible to change it.

Internally,
- `saveDHCP()` is called every time `setAddress()` is called.
   - `setAddress()` is called by `loadDHCP()` and `DHCP()`. 
      - `loadDHCP()` is only called by `begin()`. Meanwhile, `DHCP()` is meant to be called by the user application when appropriate.

As a result, I would expect this feature to also mandate any calls to `saveDHCP()` be made only from the user's application (when not running on Linux) for better control over wear and tear of non-volatile storage.

## Research links
arduino.cc has a [list of libraries devoted to data storage](https://www.arduino.cc/reference/en/libraries/category/data-storage/). Notice that these libraries are not architecturally universal. So, this feature may require a "RF24Mesh/utilities" folder for each supported architecture's NVM implementation. Not to mention the extra dependencies for libs that don't come with the Arduino's default framework (the PlatformIO package should also be compensated similarly).