nRF24/RF24Mesh #111: Two way Communication issue

darshit115, 2017-01-07:
Dear TMRh20,
I have done project of temperature sensor using RF24Mesh library in that i am sending the data from slave to master node only one way, This project works well.
Now i am doing another project using same library but in these i am doing Two Way communication in that master send command to the slave and slave will send ack signal to master. in these project while communication between 1 master and 1 slave its works but while i am adding 2nd slave node its start miss behaving after 2-3 minute and also creating problem while one slave node become child of another slave node.

can you help me to solve these issue?
Thanks and Regards,
Darshit

Avamander, 2017-01-07:
> its start miss behaving after 2-3 minute  

Please explain what does misbehavior in this context means.

> and also creating problem while one slave node become child of another slave node.

That's not a bug, that's intended to work like that. That's how you end up with a mesh.

darshit115, 2017-01-07:
Thanks for your reply.
I have created mesh network using one master and three slave node. Aim of these Mesh Network is to send command from master side to particular slave node and slave node will send ack to master that i have received data. but Problem is that at every 2-3 minute communication between master and slave nodes are dropped and connection loss remain for 1-2 minute and again it start working correct.

Inside the loop, Master side i am calling mesh.update() and mesh.DHCP() with delay(2) continuously and Slave side i am checking connection using mesh.checkConnection() and calling mesh.update() with delay(2) continuously. 

So can you suggest me is there any problem in above functionality in code. or can you suggest any better idea so mesh remains updated without loss the connection.

Thanks & Regards,
Darshit

Avamander, 2017-01-07:
You should check the connection once in a while and reconnect when
necessary on the slave nodes.

darshit115, 2017-01-07:
slave side code. is it correct?

void loop() 
{
  if(!mesh.checkConnection()) 
  {
    mesh.update();
    delay(1);
    mesh.renewAddress();
  }
  if(network.available())
  {
    //Serial.println("Data available form Mesh");
    RF24NetworkHeader header;
    payload_command payloadC;
    network.read(header, &payloadC, sizeof(payloadC));
    //Serial.println("Data readed...");
    if(payloadC.nodeId==nodeID)
    {
      if(payloadC.command==10 || payloadC.command==11)
      {
        if(payloadC.command==10)
        {
          digitalWrite(LED,HIGH);
        }
        else if(payloadC.command==11)
        {
          digitalWrite(LED,LOW);
        }
        payload_ack payloadA={nodeID,1};
        sleepTimer = millis();
        bool timeout=false;
        while(!timeout)
        {
          //Serial.println("Sending Ack...");
          if(!mesh.write(&payloadA, 'M', sizeof(payloadA))) 
          {
            if(millis()-sleepTimer>=3000)
            {
              //Serial.println("Timeout during sending Ack");
              timeout=true;
            } 
            if(!mesh.checkConnection()) 
            {
                mesh.update();
                delay(1);
                mesh.renewAddress();
          }
          else
          {
            //Serial.println("Data send ok");
            break;
          }
          mesh.update();
          delay(1);
        }
      }
    }
  }
  mesh.update();
  delay(1);
}

darshit115, 2017-01-07:
i have also try to begin the connection from slave side using mesh.begin(); but its not connecting with master. but when i reset the master then only slave connect with master.
if(!mesh.checkConnection()) 
  {
    mesh.update();
    delay(1);
    mesh.renewAddress();
      //Serial.println("In Loop checkConnection and renewAddress");
      uint16_t Add;
      if((Add = mesh.renewAddress(2000))>0)
      {
        //Serial.print("In Loop Address is renew :");
        //Serial.println(Add); 
      }       
      else
      {
        Serial.println("In Loop Failed to renewAddress");
        Serial.println("*********WDT_ENABLE**********");
        wdt_enable(WDTO_8S);
        if(!mesh.begin(CHANNEL,RF24_250KBPS,9000))
        {
            Serial.println("mesh.begin()");         
            Serial.print("In Loop Try to connect with Channel : ");
            Serial.println(CHANNEL);
        }
        wdt_disable();   
        Serial.println("*********WDT_DISABLE**********");       
      } 
  }
  else
  {
     Serial.println("In Loop Connected....");  
  }

closed 2019-05-10 by TMRh20