nRF24/RF24Mesh #112: Using NRF24 on Raspberry Pi3B on SPI1

cgpgh, 2017-01-07:
Hello,

I have a Adafruit PiTFT Plus 480x320 3.5" TFT+Touchscreen for Raspberry Pi display (https://www.adafruit.com/products/2441)  on my Pi3 and it uses a custom kernel on SPI0, I would like to use NRF24 Radio on the same Pi3. I have enabled SPI1 on Pi3 and able to list the SPI device. Bluetooth is disabled on the Pi3.

root@sh-base:~/rf24libs/RF24Mesh# ls /dev/spidev* 
/dev/spidev0.0  /dev/spidev0.1  **/dev/spidev1.0**
root@sh-base:~/rf24libs/RF24Mesh#

I have connected the NRF24 pins to SPI1 pins on Pi3 based on this https://pinout.xyz/pinout/spi, but I am unable to get the Radio to work. With out the Display Radio works fine on SPI0.

Does the library needs changes to accommodate NRF24 Radio on SPI1?

Any suggestion or help is highly appreciated. Thanks in advance.

adamdabrowski04, 2017-01-21:
Bluetooth is connected to serial port on RPI3(tx rx lines) not SPI(MOSI MISO CLK). But if you want reliable serial read [this](http://spellfoundry.com/2016/05/29/configuring-gpio-serial-port-raspbian-jessie-including-pi-3/)  I  use one additional node for communication with RPi3.  I connected arduino with RPi through serial port and it works great easpecially through USB ( bigger data buffer).

cgpgh, 2017-01-21:
My question is not with using bluetooth. My SPI0 is used by  Adafruit LCD display and I need to use the NRF24 Radio on a SPI1. The RF24 library uses SPI0 pins, how do I make it to use SPI1?

Thank you

cgpgh, 2017-02-02:
Folks any suggestion or help on getting Nrf24 radio to work on SPI1 on raspberry pi 3?

chriskelman, 2017-02-02:
Do you mean GPIO pins 10 - 14?

cgpgh, 2017-02-02:
Thank you for for the response.

In this example below, which I use.
https://github.com/TMRh20/RF24Mesh/blob/master/examples_RPi/RF24Mesh_Example_Master.cpp

** RF24Mesh_Example_Master.ino by TMRh20

#include "RF24Mesh/RF24Mesh.h"  
	#include <RF24/RF24.h>
	#include <RF24Network/RF24Network.h>

RF24 radio(RPI_V2_GPIO_P1_15, BCM2835_**SPI_CS0**, BCM2835_SPI_SPEED_8MHZ);  

It calls for SPI_CS0, in my case SPI0 CS0 is used by adafruit LCD on my Pi3. What can I use for CS for NRF24 radio?

chriskelman, 2017-02-03:
I haven't experimented with doing this yet (though I have previously done penance working out how to re-allocate SPI pins on the Arduino...)

You could try the mod below - I understand that bcm2835.h supports this (see line 816 onwards)

This should allow you to connect to CE1 (GPIO 11) instead of CE0 (GPIO 10)

RF24 radio(RPI_V2_GPIO_P1_15, BCM2835_SPI_CS1, BCM2835_SPI_SPEED_8MHZ);

https://github.com/TMRh20/RF24Mesh/blob/master/examples_RPi/RF24Mesh_Example_Master.cpp

cgpgh, 2017-02-03:
Thank you chriskelman, I will try what you suggested. Thanks again for your help.

closed 2019-05-10 by TMRh20