nRF24/RF24Mesh #115: Second RF24Mesh in the vicinity / rogue detection

rumbero71, 2017-01-28:
Hi,
coming from a WIFI-perspective, I was wondering what happens if e.g. a neighbour was to set up his own Mesh. From what I understand is that there is no such thing as a Mesh ID so I would assume that the two meshes end up interfering with one another and -although perhaps not understanding the others' data format- crippling its operation. Is that correct or did I miss something ? If correct, how can this situation be mitigated, resp. how would something like a rogue detection work ?
Anyways...great work !
Regards Markus

Avamander, 2017-01-28:
Yes, meshes on the same channels end up interfering with each other. How to detect? Well, if you could make master listen for packets that only a master node is supposed to send that would be one way. Though it would require modifications to the library itself.

rumbero71, 2017-01-29:
Ok then. So how do I limit my mesh to a dedicated channel. I tried radio.setChannel before or after mesh.begin on both Master and Slave Node besides running my "productive" mesh without those limits (is there a default channel ?) and -strangely enough- "productive" master has the NodeID of the "limited" slave, but not seeing packets, whilst the "limited" master does see the packets without having the "limited" slave's NodeID. Where is my error in thinking ?

TMRh20, 2019-05-10:
Call  ` mesh.begin(channelNumber); `

closed 2019-05-10 by TMRh20