nRF24/RF24Mesh #116: Using RF24Mesh with interrupts

estebanrolando, 2017-02-11:
I developed a gateway running in a Rpi using RF24Network, and a node with Arduino.
The gateway uses interrupts to minimize cpu utilization.
I decided to "upgrade" the system to use RF24Mesh, so I went ahead and modify the code as needed.
I added mesh.update() and mesh.DHCP() inside the interrupt handler (with RF24Network, network.update() was inside the interrupt handler code).
When I started the gateway and client, the arduino client never went beyond mesh.begin(). It was stuck trying to acquire an address.

After many tests, I changed the code in the interrupt handler from:

  	mesh.update();
  	mesh.DHCP();
	radio.whatHappened (txOk, txFail, rxOk );
	if ( rxOk )

to:
	radio.whatHappened (txOk, txFail, rxOk );
  	mesh.update();
  	mesh.DHCP();
	if ( rxOk )

and the node was able to get the address and everything started working.

Are there any recommendation on how to use mesh with interrupts? 

So far I have been doing limited testing (only one node) and it's working fine.

I also tried to move the mesh.update() and mesh.dhcp() to a separate thread, and didn´t work.

I cannot put it in the main loop, because it is blocking code. I can refactor the code but first I want to know what is the recommendation, if any.

Thanks! And great work with this libraries!!!

Esteban

TMRh20, 2017-02-22:
It gets complicated, and can't remember all the details, but the best example I can give is RF24Gateway. I have examples using interrupts with and without a ncurses GUI for RF24Mesh. 

Basically the radio will switch tx/rx during update functions so any interrupts need to be blocked while accessing the radio from main loop. I can't remember how it all works exactly and the examples are a bit of a mess, but it does work lol.

estebanrolando, 2017-02-23:
I´ll check that. So far it has been working ok with the change I mentioned before...

Thanks!

closed 2019-05-10 by TMRh20