nRF24/RF24Mesh #118: 250kbps Mesh: Long time to connect

rumbero71, 2017-03-10:
I am trying to build up a mesh of sensors where battery consumption and -to some lesser extent- range are of the essence. Mains-powered repeating nodes are possible but to maintain a decent WAF, there should be few and they should be hidden.
First of all I set up the mesh with 1MBps. All went well but range was quite poor, about 4 meters with and 5 meters without a wall. Reducing the speed to 250KBps extended the range to a more or less acceptable level, about 7 meters including walls. **However, I realized that the nodes took about a minute to connect to the mesh, even when placed right besides the master node.** Once connected all was fine.
Is there any way to reduce the time needed for connect, any best practices for doing this ?

chriskelman, 2017-03-11:
I have had variable success with these devices - when they work they are brilliant.

I find my mesh network (multiple Arduino based sensor stations with nRF24+aerial modules) will manage on Max Rate up to around 70m with immediate connection to the (Raspberry Pi) base station, even when via a 'relay' station. I have used YAGI antennae on some more distant stations with good results.

However at other times, seemingly randomly, various sensor stations will either not connect at all or take a very long time to handshake. Quite often on my system, the base-station network node allocation does not work properly and several sensor stations will end up with the same (base-station) ID, which makes accurate data recording difficult.

I have used various sources of nRF24 modules, most recently from a supplier who guarantees that they use original Nordic chips.

Maybe the combination of counterfeit and original chips is the problem - although I have tried using all original chips without much improvement in reliability.

So, I wish  you the best of luck - the technology is tantalising but I have not convinced myself that it is reliable enough to persevere with on my next project.

Hopefully there is someone much smarter than me in this group with some suggestions for my (and your) problems!

Avamander, 2017-03-19:
Well the easiest way to ensure better transmission distance is using high-power transceivers. You can also try other channels to avoid interference from other 2.4GHz signals and place more nodes between others to repeat. There's not much else you can do. Lowering transmission speed might not be the best as you can see.

closed 2019-05-10 by TMRh20