nRF24/RF24Mesh #119: RF24Mesh Max Nodes

CJTalbot87, 2017-03-19:
Question: Am I correct in assuming that a RF24Mesh Master can only communicate with 254 sub nodes and that the underlying RF24Network library is capable of 3,125 per channel?

Avamander, 2017-03-19:
Indeed, though, well, good luck not congesting your receiver with 3 125 nodes or even 255.

andrewringler, 2017-06-23:
Do you know how many nodes your software does support? What have users been able to achieve?

Avamander, 2017-06-23:
Given enough memory for the address table and that your nodes don't exceed the theoretical maximum bandwidth of 2Mbits (~-30% in real world) you can have 255 nodes with the current code. There are too many factors in play to say a specific number what you'll actually be able to reach with your setup. Trial and error is the only way to know.

OldSurferDude, 2018-03-13:
I just finished setting up a 12 node (including master) mesh.  I modified the master code a bit to check to see if any nodes are being lost (no contact after 10 seconds).  After 24 hours, two nodes seem to have gone permanently off line but some nodes come and go.    

So my trial and error tells me 10 nodes.

Note that I'm using the cheapest arduino nanos, nRF24 modules, and USB chargers I can find.  Thus, that they all initially work is pretty amazing.

(I'll be asking questions when I get my documentation in order and better understand what I have)

TMRh20, 2019-05-10:
IMHO it all depends on the traffic generated by the nodes. For example, 1 node sending @ about 25-30KBPS will nearly max out the system, while a mesh with 25 nodes, each sending data every minute would be fine.

closed 2019-05-10 by TMRh20