nRF24/RF24Mesh #120: Redundancy - secondary master node?

AdrieK, 2017-04-28:
I can't seem to find a solution on how to handle a master node down situation. Could there be a secundary master that takes over this role if for any reason node 0 goes offline? (Or will the network just keep working?)

Avamander, 2017-04-28:
You probably have to create that solution yourself. The mesh will not work without a master.

AdrieK, 2017-04-28:
Thanks for your quick reply. I am building a network usinf RF24network until I discovered this issue, hoped that it was build in in RF24Mesh. I'll do some thinking, and some coding. Maybe 'node 1' can reconfigure itself if node 0 is missing? Or a node can be designated as secundary - sychronising with master. I will post here if I succeed (or have further questions).
(Ideally RF24Network should have this built in already?)

Avamander, 2017-04-28:
I haven't had the time but I have definitely thought about that. You can try initialising the libraries with different nodeID once a node detects master has gone down. But you also have to handle fallback to being a node and cases where two masters appear.

danhajduk, 2017-04-29:
I'm working on smart-home library that is using RF24Mesh for communication.
i have the same problem when the main hub goes down the whole structure is falling apart
I did not dig in the mesh library but do you have any idea how to approach the issue?
my library is on [https://github.com/danhajduk/SmartHome](url)

TMRh20, 2017-04-29:
I was considering previously a scenario where two masters listen on separate channels in the same area. Then all nodes get programmed to hop after a set period of connectivity loss and/or retry main channel. Very simple to program onto the nodes in different ways, but I haven't  really tried it. The master would also need to remove nodes after a set time of no communication etc...

If an individual master is left offline, nodes are normally programmed to renew their address and re-form the structure after a set interval, once a master is back online.

closed 2019-05-10 by TMRh20