nRF24/RF24Mesh #124: Dynamic network

smiron, 2017-09-08:
Hi,

Is it possible to configure a mesh that dynamically determins the router?
Essentially any node would have the possibility to be both router and child.

So every node will:
1. Try connect to mesh
2. If connection successful -> continue normally
3. If connection fails -> become the router

Thanks

Avamander, 2017-09-08:
Sure, you can programmatically reinitialise the library to be a master, but you have to do it yourself.

smiron, 2017-09-09:
Is there any way of making sure that we have only 1 router?

For example if 2 nodes start at the same time, they will both have mesh.begin() == false so they will both become routers.

Avamander, 2017-09-09:
Do what you described, try to connect if that fails become host. Race conditions (where two meshes are created) will always be possible, but you can fix the mesh if they happen, for example if you have two masters it's possible that you'll start seeing nodes sending one master messages that aren't in that master's node list and that can be detected easily (do a nodeID lookup).

closed 2019-05-10 by TMRh20