nRF24/RF24Mesh #125: Problem with other SPI modules and interrupts

obdevel, 2017-09-20:
I can successfully communicate between two Arduinos using RF24Mesh and it is generally very stable. However, when I introduce a CAN bus module on the same SPI bus I get problems.

The CAN bus module uses an interrupt to signal the arrival of new data and there seems to be an unpredictable timing issue with RF24/Network/Mesh and access to the SPI bus. Clearly, I can't control when the ISR will be triggered, and everything is mostly ok. However, every so often the radio will stop responding, i.e. network.read() returns nothing, or sometimes message type 0, and 0 in all other header fields, i.e. a blank message. Once this happens, no more radio messages are received by .available(), .peek() or .read(). But I do continue to receive CAN frames on the SPI bus.

The ISR just reads the next message frame from the CAN module (using SPI) and buffers it. I have to use an interrupt because polling is just not quick enough and messages might be dropped.

Does the library expect exclusive access to the SPI bus ? I don't know enough about SPI to know whether it has the concept of atomic transactions.

I am happy to debug if you can give me somewhere to start ? Too much code to post an example, and you'd need a CAN module to reproduce !

Thanks

obdevel, 2017-09-20:
Seems this is a limitation of SPI. Interrupts are a no-no if multiple slave devices are present on the bus.

RF24 seems amenable to using SoftSPI to create a separate 'virtual' SPI bus, so I'll give this a try and report back. SoftSPI is implemented in this library: https://github.com/greiman/DigitalIO

TMRh20, 2017-09-20:
I'm thinking you would need both libraries to support interrupts to handle this. If the CAN data could be handled outside of the ISR it might work. I would suggest an alternate SPI bus using software or the UART.

obdevel, 2017-09-20:
Thanks for the quick reply although I think we may have crossed. Fortunately your RF24 library seems amenable to SoftSPI so I'll try that.

Unfortunately, CAN frames can arrive at such a rate that polling in loop() (as well as doing other useful work) would mean missed/dropped frames. The CAN hardware doesn't buffer sufficiently either, and as it's a monitor application, I can't filter CAN frames at the h/w level.

Thanks again for the support.

chriskelman, 2017-09-21:
I too have found the soft SSPI library to be useful in these situations - I have had to use it to support non-compliant SSPI devices (such as SD card readers) and found that the nRF24 works well using the SSPI interface - while the SD reader hogs the nominal SSPI pins.

obdevel, 2017-09-24:
So ... I seem to have an explanation, and almost a resolution. The other library doesn't implement SPI transactions, so interrupts from the other device were not being masked during RF24 method calls, and thus it was executing SPI bus calls during RF24 transactions. I'm not surprised it was trampling all over the bus !

I've emailed the author and have quickly hacked the necessary changes into his code. I've also made a call to SPI.usingInterrupt() in my setup() function. That interrupt should now be masked during RF24 bus transactions.

Looks ok so far ... but I'll update here later for posterity.

closed 2019-05-10 by TMRh20