nRF24/RF24Mesh #126: Are DHCP request forwarded by nodes?

Exormeter, 2017-09-21:
Hello, I'm experimenting with your RF24Mesh library and I am very thankful for your work. 
My question is,  if a node in the network is in range of another node that is currently not in the network, will the node in the network pick up the DHCP request form the outside node an forward to the master node? 

I an currently experimenting with a Raspberry Pi as my master node and several Arduino nano node for the mesh network. While connecting with the master node works always flawless, if I try to make a node "chain" so that the master is not in range, the new node does connect to the network.

I'm using the example code "RF24Mesh_Example_Master" on the Raspi as my Master and "RF24Mesh_Example.ino" on the Arduinos (nodeID are different of course).

TMRh20, 2017-09-21:
Yes they should be forwarding the requests. You could try editing Rf24mesh-config.h and reduce the MESH_MAX_CHILDREN to 2 or so, on the master node, which should force them to link up differently. This should demonstrate whether dhcp routing etc is working at all.

Exormeter, 2017-09-21:
Thank you for your reply. That DHCP is forwarded is great news for my project. 
About the MAX_CHILD_NODE constant, I already played with it and set it to other values, but it seems to me that it doesn't make a difference. I set it for example to 1, but the master node still accepted 5 children. It feels like the Mesh library is not working correctly in my setup and is falling back to the RF24Network lib, where 5 children is  standard. I will continue my tes tomorrow, probably just a problem on my end.

Exormeter, 2017-09-22:
Hmmm, I still have the same problem as yesterday. I feel like I do something wrong. I have 6 Arduino node plus the Raspberry master node, but only 5 will get adresses. The sixth node never get an address. Also, I installed the libs with `sudo make install -B` and uncommented the debug defines, but I don't get any debug output. Only the radio details at the start of the master node.

Tryied both the master branch and the testing branch with no luck.

Exormeter, 2017-09-22:
I am an idiot, I configured the RF24Mesh_config after installing it by changing the values in the file in the include directory. If course this wouldn't work. After playing with the MAX_CHILD_NODE, it seems to work fine. Is it correct that if I set MAX_CHILD_NODE to for example 3, the number of accepted child are in fact 4? This is the observation if have made.

Also, if a node goes down due to power, will the used IP get relased after a certain time or does the master node has to check in a periodic time if all nodes are still online?

TMRh20, 2017-09-22:
Yes, master accepts max_child_nodes + 1 while other nodes use max_child_nodes. The master does not remove nodes automatically, only as requested.

Exormeter, 2017-09-28:
Your library is working really good after playing with it for some days. I made a few changes to it so the NodeID is now allocated automaticly using NodeID 255 as the default ID. Seems to be working.
Only thing I needed to change in the Network layer that is that the new message Type is forwarded.

closed 2019-05-10 by TMRh20