nRF24/RF24Mesh #131: Undefined reference to...all methods used in example program

Rotario, 2017-10-06:
Hi,
I've been using the RF24 library for a while now and it works brilliantly. However, when trying to use the mesh library I get the following errors while trying to compile the example programs:

```
g++  -Wall -I../ -lrf24-bcm -lrf24network -lrf24mesh RF24Mesh_Example.cpp -o RF24Mesh_Example
/tmp/ccGo39kA.o: In function `main':
RF24Mesh_Example.cpp:(.text+0x14): undefined reference to `RF24Mesh::setNodeID(unsigned char)'
RF24Mesh_Example.cpp:(.text+0x24): undefined reference to `RF24Mesh::getNodeID(unsigned short)'
RF24Mesh_Example.cpp:(.text+0x48): undefined reference to `RF24Mesh::begin(unsigned char, rf24_datarate_e, unsigned int)'
RF24Mesh_Example.cpp:(.text+0x54): undefined reference to `RF24::printDetails()'
RF24Mesh_Example.cpp:(.text+0x60): undefined reference to `RF24Mesh::update()'
RF24Mesh_Example.cpp:(.text+0x64): undefined reference to `__millis'
RF24Mesh_Example.cpp:(.text+0x8a): undefined reference to `__millis'
RF24Mesh_Example.cpp:(.text+0xb4): undefined reference to `RF24Mesh::write(void const*, unsigned char, unsigned int, unsigned char)'
RF24Mesh_Example.cpp:(.text+0xcc): undefined reference to `RF24Mesh::checkConnection()'
RF24Mesh_Example.cpp:(.text+0xf4): undefined reference to `RF24Mesh::renewAddress(unsigned int)'
RF24Mesh_Example.cpp:(.text+0x122): undefined reference to `__msleep'
/tmp/ccGo39kA.o: In function `__static_initialization_and_destruction_0(int, int)':
RF24Mesh_Example.cpp:(.text+0x176): undefined reference to `RF24::RF24(unsigned short, unsigned short)'
RF24Mesh_Example.cpp:(.text+0x1a6): undefined reference to `RF24Network::RF24Network(RF24&)'
RF24Mesh_Example.cpp:(.text+0x1de): undefined reference to `RF24Mesh::RF24Mesh(RF24&, RF24Network&)'
/tmp/ccGo39kA.o: In function `RF24::~RF24()':
RF24Mesh_Example.cpp:(.text._ZN4RF24D2Ev[_ZN4RF24D5Ev]+0x14): undefined reference to `SPI::~SPI()'
collect2: error: ld returned 1 exit status
Makefile:39: recipe for target 'RF24Mesh_Example' failed
make: *** [RF24Mesh_Example] Error 1
```
I've git cloned and then sudo make, sudo make installed both the RF24 network and RF24 mesh libraries. The RF24 network examples compile fine.
Would you have any ideas to fix this please?
Thanks in advance.

Avamander, 2017-10-06:
Clearly it can't find RF24Network and RF24 either, so clean your build environment up and try again.

Rotario, 2017-10-06:
Well, RF24Network and RF24 compile just fine - I installed RF24mesh the same way but it's not found. I think it's something to do with the makefile - which I've messed around with. The RF24mesh.h file is installed correctly in /usr/local/include.

Avamander, 2017-10-06:
```
RF24Mesh_Example.cpp:(.text+0x54): undefined reference to `RF24::printDetails()'
RF24Mesh_Example.cpp:(.text+0x64): undefined reference to `__millis'
RF24Mesh_Example.cpp:(.text+0x8a): undefined reference to `__millis'
RF24Mesh_Example.cpp:(.text+0x122): undefined reference to `__msleep'
RF24Mesh_Example.cpp:(.text+0x1a6): undefined reference to `RF24Network::RF24Network(RF24&)'
```

This indicates literally everything being missing, include all the header files that are needed properly.

closed 2018-01-08 by Rotario

mcer12, 2018-09-12:
@Rotario I'm having the same problem, how did you fix it?