nRF24/RF24Mesh #135: Running RF24Mesh with CRC8

MAKOMO, 2017-10-30:
In case payloads are rather small, using CRC8 instead of the CRC16 might be an option to further reduce payloads by another byte.

However, 

```
mesh.begin(/* channel */ RADIOCHANNEL, /* speed */ RF24_250KBPS);
radio.setCRCLength(RF24_CRC_8); 
```

seems to fail as mesh.begin calls radio.begin(), setting the CRC to the 16bit default, followed by renewAddress(), called before the CRC is set to 8bit after the mesh.begin. 

Shouldn't CRC be a third argument to mesh.begin to allow mesh networking using CRC8 instead of CRC16, or does this not make any sense?

MAKOMO, 2017-10-30:
Oh, and it seems that this initial renewAddress() called from within mesh.begin() is running using the default PAlevel (whatever that is) and the caller has no chance to specify a different one.

TMRh20, 2020-08-16:
I've decided against putting CRC configuration as an option in mesh.begin, and pretty much the same with PALevel.

Technically any radio setting other than the channel and dataRate can be configured prior to calling `mesh.begin();`, just call `radio.begin();` first. I almost want to remove the options from RF24Mesh and just have them set prior to calling mesh.begin(),  so its all the same because it goes on forever... how many options to allow in the constructor?

closed 2020-08-16 by TMRh20