nRF24/RF24Mesh #138: ATtiny85 + NRF24L01 Node need other node to be a "Starter" to run.

William1993, 2017-11-16:
Hi, I have an issue with the node that using Attiny85 as a microcontroller. It works but only when there is another node (or slave) run with it at the beginning. I do not sure because of hardware or software problem. Thus, below are my circuit connection and codes.

Let's define some names to making thing more easier, I will call:
- "ATtinyNode" for the node that using Attiny85 and NRF24L01 as a slave with nodeID = 1.
- "NanoNode" for the node that using Arduino Nano and NRF24L01 as another slave with nodeID = 2.
- "NanoMaster" for the node that using Arduino Nano and NRF24L01 as the master.

For the ATtinyNode, I follow the circuit below (from  [https://create.arduino.cc/projecthub/arjun/nrf24l01-with-attiny85-3-pins-74a1f2]()) to connect the ATtiny and NRF24 together using 3V power source:
[image]

For the NanaNode and NanoMaster,  I follow the circuit below from the same website as above, I use Arduino Nano instead of Arduino Uno as shown in the circuit diagram. However, all pins are connected the same.
[image]

**Code for ATtinyNode:**
```
#define CE_PIN 3
#define CSN_PIN 3 //Since we are using 3 pin configuration we will use same pin for both CE and CSN

#include "RF24.h"
#include "RF24Network.h"
#include "RF24Mesh.h"

/**** Configure the nrf24l01 CE and CS pins ****/
RF24 radio(CE_PIN, CSN_PIN);
RF24Network network(radio);
RF24Mesh mesh(radio, network);
/**
   User Configuration: nodeID - A unique identifier for each radio. Allows addressing
   to change dynamically with physical changes to the mesh.

   In this example, configuration takes place below, prior to uploading the sketch to the device
   A unique value from 1-255 must be configured for each node.
   This will be stored in EEPROM on AVR devices, so remains persistent between further uploads, loss of power, etc.

 **/
#define nodeID 1

uint32_t displayTimer = 0;

struct payload_t {
  unsigned long ms;
  unsigned long counter;
};

void setup() { 
  // Set the nodeID manually
  mesh.setNodeID(nodeID);
  // Connect to the mesh
  mesh.begin();
}

uint32_t ctr = 0;

void loop(){
  mesh.update();

  // Send to the master node every second
  if (millis() - displayTimer >= 1000) {
    ctr++;
    payload_t payload = {millis(), ctr};

    // Send an 'M' type message containing the current millis()
    if (!mesh.write(&payload, 'M', sizeof(payload))) 
    {

      // If a write fails, check connectivity to the mesh network
      if ( ! mesh.checkConnection() ) 
      {
        mesh.renewAddress();
      }
    }
    displayTimer = millis();
  }
}
```

**Code for the NanaNode** is the same except CE_PIN and CS_PIN changed to 7 and 8 respectively, the nodeID also changed to 2.

**Code for NanoMaster:**

```
#include "RF24Network.h"
#include "RF24.h"
#include "RF24Mesh.h"
#include <SPI.h>
//Include eeprom.h for AVR (Uno, Nano) etc. except ATTiny
#include <EEPROM.h>

/***** Configure the chosen CE,CS pins *****/
RF24 radio(7,8);
RF24Network network(radio);
RF24Mesh mesh(radio,network);

uint32_t displayTimer = 0;
struct payload_t {
  unsigned long ms;
  unsigned long counter;
};

void setup() {
  Serial.begin(115200);

  // Set the nodeID to 0 for the master node
  mesh.setNodeID(0);
  Serial.println(mesh.getNodeID());
  // Connect to the mesh
  mesh.begin();

}

void loop() {    

  // Call mesh.update to keep the network updated
  mesh.update();
  
  // In addition, keep the 'DHCP service' running on the master node so addresses will
  // be assigned to the sensor nodes
  mesh.DHCP();
  
  
  // Check for incoming data from the sensors
   while (network.available()) {
    RF24NetworkHeader header;
    payload_t payload;
    
    switch(header.type){
      // Display the incoming data values from the sensor nodes
      case 'M': 
        network.read(header, &payload, sizeof(payload));
        Serial.println("\nReceived packet: ");
        Serial.print("Send OK; couter: ");
        Serial.print(payload.counter);
        Serial.print(" at ");
        Serial.print(payload.ms);
        break;
      default: network.read(header,0,0); Serial.println(header.type);break;
    }
  }
  
  if(millis() - displayTimer > 5000){
    displayTimer = millis();
    Serial.println(" ");
    Serial.println(F("********Assigned Addresses********"));
     for(int i=0; i<mesh.addrListTop; i++){
       Serial.print("NodeID: ");
       Serial.print(mesh.addrList[i].nodeID);
       Serial.print(" RF24Network Address: 0");
       Serial.println(mesh.addrList[i].address,OCT);
     }
    Serial.println(F("**********************************"));
  }
}
```
The problem is:
When the NanoMaster and the ATtinyNode are on (NanoNode is off), the Master recieve no payload.

[image]
When the NanoMaster and the NanoNode are on (ATtinyNode is off), they work ok, the Master can receive the payload.
When the NanoMaster and the ATtinyNode are on (Master recieve no payload as stated above) and then a few moment later, the NanoNode is on as well. The Master now receive both payloads from ATtinyNode and NanoNode. They both counts from 1 even the ATtinyNode was on before the NanoNode.

[image]
Then the NanoNode is turned off and the Mater still can receive the payload for ATtinyNode.

[image]
Thus, it looks like the ATtinyNode needs a startup from another node to be able to connect to Mesh. Have I done anything wrong? Please let me know if there is something wrong and how can I fix this problem. Thank you!

positron96, 2018-08-02:
Hello, sorry I cannot help you, but how did you manage to fit the code into 8k memory of attiny85?

TMRh20, 2022-03-19:
See https://github.com/nRF24/RF24Mesh/issues/200 , most likely the same issue

closed 2022-03-19 by TMRh20