nRF24/RF24Mesh #141: 24Mesh on an ATmega328P: Slow/Freezing?

Gaz83, 2018-02-01:
Trying the 24Mesh on an ATmega328P and I seem to be running in too an issue.

Firstly, if i try running the RF24-Getting Started Sketch, everything works fine.

When I then try to run RF24Mesh_Example, on the serial monitor I see 

**Connecting to the mesh...**

On my Master device I see no new devices join.

So I tried looking at the radio settings using the printDetails and this is where I spot an issue.
In the serial monitor I get the connectiong to Mesh and then about 5 minutes later I get the radio details.

My setup looks like this

void setup()
{  
  Serial.begin(115200);
  
  printf_begin();
  // Set the nodeID manually
  mesh.setNodeID(nodeID);
  // Connect to the mesh
  Serial.println(F("Connecting to the mesh..."));
  mesh.begin();  

    radio.printDetails();
}

Avamander, 2018-02-01:
So let's start from the beginning, let's make sure if you have your
environment all up to date. Make sure all the RF24 libraries (RF24,
RF24Network and RF24Mesh) are the latest versions, then make sure your
Arduino IDE is on the latest stable version.

After that is done, please confirm your hardware is properly connected, bad
connections can cause various issues, voltage drop, noise and high
resistance (ribbon cables are notorious for this) which can cause issues
with, you guessed it, the proper functioning of the library stack. When
that is checked, solder(!) a 10000uF electrolytic capacitor to directly on
top of the nRF24L01+ module.

Now, the next step would be verifying that all functionality of the modules
are working, RF24Mesh(/Network) demands pretty much every feature of the
chips to work. To test this you should start with the PingPair example
(part of RF24), it should work really really stable (no more than 1% loss)
over a reasonable timeframe. Then move on to the PingPair ACK (RF24)
example, it too should work as stable as the previous one. After that move
on to the PingPair DYN (RF24 again) example, it must work just as well. The
next test would be RF24Network's examples, the most basic one should
suffice to make sure that part works too. You can also try the Transfer
example and write those results down, nice way to compare different modules
IMHO.

Now if after all that, everything has worked perfectly, when you have to
wait 5min for those to work or have to wiggle some wires it ain't
perfectly, then report back here.

Gaz83, 2018-02-03:
Ok, first the setup I have

2 Nodes, Mega2560 RS and a custom board with the ATmega328p. All wiring on the custom board is soldered, buzzed out a million times, voltages checked and I am happy. I also added the capacitors to modules on both.

I could not see the PingPair Example you mentioned so I started with PingPair ACK. Both boards failed to send. Both boards failed to receive. Obviously I had them at opposite roles.

I then tried PingPair dyn.

Custom Board received everything
Mega2560 kept saying sending failed.

Role Swap

Custom Board kept saying sending failed.
Mega2560 received nothing
Replaced the module on the Mega2560, still the same.

Avamander, 2018-02-03:
The most basic pingpair example is the GettingStarted one, should've clarified that. Please report back if that works.

Gaz83, 2018-02-03:
Ok just tried that.

Custom Board set to Radio 1
Mega2560 set to Radio 0

Test 1 - Custom Board Set to Send, Mega2560 set to respond
Seeing some sends as fail but most are working, with Mega2560 responding back

Test 2 - Custom Board Set to Respond, Mega2560 set to Send
Custom Board gets message and responds, Mega2560 always reports failed send

Avamander, 2018-02-03:
Did you complete the hardware checkup and improvement?

Gaz83, 2018-02-03:
Yes, I have swapped the Mega2560 with a Nano. Repeated the above and still the same result. I then swapped the modules and still the same result.

Avamander, 2018-02-03:
If that's what you say then it's the modules being fakes. You can't use RF24Mesh properly, but you can try squeeze out something with just RF24 and no auto-ACK and dynamic payloads.

Gaz83, 2018-02-11:
Sorry for the late reply. I have ordered some more modules from Itead Studio. Read in a few places that these are genuine. Once they arrive I will repeat what I have done and report back.

Gaz83, 2018-02-17:
Ok, so the modules have arrived and I did another round of testing.
Again, I am using a NANO on a breadboard and my custom board.

Test PingPair_Ack
both boards work perfect

Test PingPair_dyn
Custom Board works perfect.
Nano has some issues with sending but is fine receiving. I swapped the modules in case I had a faulty one but this did nothing.

Test - Network 
Both boards work perfect

Test - Mesh
Both boards have the same issue as I originally posted.

Avamander, 2018-02-20:
Yeah, issues with dynamic payloads indicates counterfeit modules, nothing can be done there except not use dynamic payloads.

closed 2018-02-20 by Avamander