nRF24/RF24Mesh #144: nRF24l01 mesh network with four sensor nodes and one master

ashok-singamreddy, 2018-03-07:
Hardware used:
1st node have 2 ultrasonic sensors, one push button momentary switch, arduino uno and a nRF24l01
2nd node have only one push button momentary switch, arduino uno and a nRF24l01
3rd node also have push button momentary switch, arduino uno and a nRF24l01
4th node have one stepper motor, arduino uno and a nRF24l01
Master node: this node have a single nRF and arduino uno to receive and transmit the data from and to the other nodes.

Application:
                   The master node will be connected to the COM port of PC. Each node in the network has to to send the data every time to the master except 4th node. 1st node will send two distance values and a switch state(1 or 0). 2nd and 3rd  nodes will send switch state every time to the master. 4th node will receive the stepper motor angle value sent by the master. The master node will receive the all nodes data and here i am writing these values to the serial monitor (COM port). The master is supposed to receive serially one after another like 1st node, 2nd node and then 3rd node every time, but it is not happening at times. Sometimes the nodes fails to send data to master. On the COM port of the master has to be like this 51(1st ultrasonic sensor value),51(2nd ultrasonic sensor value),0(switch in the 1st node),0(2nd node switch),0(3rd node switch).
On the serial monitor it supposed to be like this for infinite time 
5151000
5151000
5151000
5151000
5151000
5151000
But sometimes it is missing some node value. In result i am getting the data on serial monitor like
51(1st ultrasonic sensor value),51(2nd ultrasonic sensor value),0(switch in the 1st node),0(3rd node switch).
515100
0051510
5151000
51510
How can I get the data with out any loss of data from all nodes?
Kindly Expecting a valuable reply soon,

Master code:
`
#include <SPI.h>
#include <EEPROM.h>
#include <Wire.h>

#include "RF24Network.h"
#include "RF24.h"
#include "RF24Mesh.h"

RF24 radio(7,8);
RF24Network network(radio);
RF24Mesh mesh(radio,network);

struct payload_from_slave {
  char buf[20];
   int switchh; 
}payload={0,0};

struct payload_from_slave1 {
int motor;
int distance;
int distance1; 
}payload1={0,0,0}; 

struct payload_t {
  unsigned long ms;
  unsigned long counter;
};

int Node[10], nodeID;
uint16_t value;
uint32_t displayTimer = 0;
String Temp[10], myString[10]; 

void setup() {
  Serial.begin(115200);
  mesh.setNodeID(0);
  Serial.println(mesh.getNodeID());
  mesh.begin();
}

void loop() {    
  mesh.update();
  mesh.DHCP();
    
  if(network.available()){
    RF24NetworkHeader header;
    network.peek(header);
    switch(header.type){
      // Display the incoming millis() values from the sensor nodes
      case 20: network.read(header, &payload1, sizeof(payload1));
                Serial.print(payload1.distance);
                Serial.print(payload1.distance1);
                Serial.print(payload1.motor);
                break;  
      case 30: network.read(header, &payload, sizeof(payload));
                Serial.print(payload.buf);
                Serial.println(payload.switchh);
                break;                        
      default: network.read(header,0,0); 
               Serial.println(header.type);
               break;
    }

  
  }

  
   if (millis() - displayTimer > 1000) {

    for (int i = 0; i < mesh.addrListTop; i++) {
      payload_t payload = {millis(), 90};
      if (mesh.addrList[i].nodeID == 3) {  //Searching for node one from address list
      }
      RF24NetworkHeader header(mesh.addrList[i].address, OCT); //Constructing a header
      network.write(header, &payload, sizeof(payload));

    }
    displayTimer = millis();
  }

}
  `

OldSurferDude, 2018-03-14:
Could be that the data being sent is overwhelming the network.  Some of the data is not getting to the master.  I am using the example mesh of tmrh20 gave (with a few modifications).  I found that I was regularly getting "Send Fail, test OK"  Data from some of the nodes were not arriving at the master.  The nodes appeared to be not functioning, "MIA" as I called it.

Here's the experiment I tried on my 12 node network.  On three of my nodes (I was going to do all 11 but life interfered) I increased the interval between data sends 5x.  I saw significant improvement in successful data sends.

My theory is that there were now fewer packet "collisions" (see https://computer.howstuffworks.com/ethernet8.htm )  In a dense data flow, there will be more collisions. These packets will be resent, making the flow even more dense.   

It also could be that you're transmitting on the same channel as a WiFi network and colliding with that data flow.

ashok-singamreddy, 2018-03-14:
Thanks for the replay.
"Here's the experiment I tried on my 12 node network. On three of my nodes (I was going to do all 11 but life interfered) I increased the interval between data sends 5x. I saw significant improvement in successful data sends."
I will try with this.

OldSurferDude, 2018-03-15:
I had another thought.  There could be an issue with the power setting of the nRF24 module.  There are four levels (if I remember correctly).  Obviously, too low, but also too high can over drive the receivers (like the distortion you get when you turn up your stereo too loud.)

closed 2019-05-10 by TMRh20