nRF24/RF24Mesh #145: Need some help Using the EEPROM

OldSurferDude, 2018-03-16:
I have built a 12 node (including master) network see: [OSDs Mesh](https://drive.google.com/open?id=1g0kyxJGDJmFJ2JhrkDX63Uh84jOvYPj-)

Most of my non-master nodes are identical except for the nodeID.  Therefore, I'd like to put the nodeID into the EEPROM, so when I load a new version I don't have to modify the code for identical nodes.

The first challenge is that I don't know if the libraries use the EEPROM, but I suspect they do because it's one of the includes.  If so, how do I find out where so I can find a spot where I can use it?

The second part is going to be tough because nodeID is a #define and that "#define nodeID n" must be after #include "RF24Mesh.h"

Thanks in advance

TMRh20, 2018-03-16:
RF24Mesh used to use EEPROM to store nodeIDs, but its now left up to the user to configure. 

It is pretty straightforward to load the nodeIDs into EEPROM and retrieve the values. See https://www.arduino.cc/en/Tutorial/EEPROMWrite and https://www.arduino.cc/en/Tutorial/EEPROMRead

You would have to go through each node, loading the nodeID into an eeprom memory location.
Then the #define for nodeid can be removed, and EEPROM read in setup, right before the nodeID is assigned.

OldSurferDude, 2018-03-17:
Your answer was as I had hoped and it works fine.  

Now on to power level testing.  I have just noted that changing the output power does not perceptibly change the power consumption.

closed 2018-03-17 by OldSurferDude