nRF24/RF24Mesh #147: node stops communicating

OldSurferDude, 2018-03-24:
This is my observation;

A node will not be able to send a message   Sometimes it recovers after the "mesh.checkConnection()", which returns true.  

It is when that function returns false that the problem shows itself (sometimes).  When it does I have the node execute a "mesh.releaseAddress()", ignoring the results (usually false) it is followed by a "mesh.renewAddress()".  Then it will be stuck in the loop trying to release and renew, to no avail.

While it is stuck in the loop, I have restarted the master a couple of times, to no avail.  I have tried sending packets to the node.  (Is there a broadcast packet?)

I'm looking for a sequence to execute to ensure that the node will reconnect.  Power cycling would be a really crappy solution.  Should I put in "mesh.checkConnection()" before the release ... after?  Should I put in a "mesh.begin(networkFrequency,networkSpeed)" someplace?  Is their a "mesh.end()"?

You can find my documentation at [this link](https://drive.google.com/open?id=1g0kyxJGDJmFJ2JhrkDX63Uh84jOvYPj-)

Thanks for your help.

TMRh20, 2018-03-25:
Theoretically, there is no sequence that should work, however I would be interested if power cycling does work. If a node cannot renew, it should mean that other nodes aren’t responding quick enough or that there are no nodes available with free addresses, or the node failed in an unacceptable way. Make sure mesh.update is called as often as possible on other nodes, and maybe try physically moving the failed node around, to see if it is really stuck.

OldSurferDude, 2018-03-25:
What's the blush emoticon?  Well sort of.  It is true that I had a broken wire. ⭕️ After the wire was resoldered, I was still able to put the node into a non-connecting mode by sending rapid fire packets to it from the master.  I was able to see it recover by sending a few packets from the master slowly.

But, after I rewrote my ping routine, I was unable to get it to fail.

I had a lot of grief from the ping program I wrote as I was getting false for network.available() after the master sent the ping packet.  (The node had replied nicely).  I ended up putting mesh.update() just before the network.available() and that resolved that issue.  

The master side is essentially send out a byte and wait for a response with a software timeout.  The node side is note that a ping packet came in and send the compliment of the byte back.  3mSec

Thanks for the tip on the mesh.update!

closed 2018-03-25 by OldSurferDude