nRF24/RF24Mesh #151: Conflict with Adafruit SHT31 D

shinn16, 2018-09-17:
I am having an issue with using the API in conjunction with the Adafuit SHT31-D temperature and humidity sensor (https://www.adafruit.com/product/2857)(API: https://github.com/jacobstim/Adafruit_SHT31D). Whenever a call to read the humidity from the sensor is made, it prevents the node making that call from relaying any information from any other node, effectively breaking the mesh. That node is, however, able to still talk to its parent node, the issue only impacts relaying messages from its children. I am not much of a hardware guy, but I tried switching the I2C address of the sensor. This had no effect. The sensor has the ability to get both temperature and humidity data, only the humidity data call causes this issue. Any ideas as to what I can do to fix this or will I just have to replace the sensor?

shinn16, 2018-09-17:
I forgot to mention my other hardware, I am running this on an Arduino Uno. I am using two ultrasonic sensors as well.

TMRh20, 2018-10-01:
Sounds like a weird issue.  I don't really have any suggestions.

Avamander, 2018-10-01:
How much space is used by the sketch in RAM and ROM?

shinn16, 2018-10-01:
Sketch uses 18518 bytes (57%) of program storage space. Maximum is 32256 bytes.
Global variables use 1207 bytes (58%) of dynamic memory, leaving 841 bytes for local variables. Maximum is 2048 bytes.

Avamander, 2018-10-01:
This could be some weird out-of-bounds write messing up RF24Mesh's data structures. I'd try some other library to read the sensor. Also possibly try to reimplement the humidy reading function using only the wire (i^2c) library.

igormp, 2018-10-06:
After skimming through the library's code, I've found some uses of the `delay()` function (as seen [here](https://github.com/jacobstim/Adafruit_SHT31D/blob/bcc8e23f50d8d6da7ab9d12e00bd67af2e8ccc60/Adafruit_SHT31.cpp#L343) and [here](https://github.com/jacobstim/Adafruit_SHT31D/blob/bcc8e23f50d8d6da7ab9d12e00bd67af2e8ccc60/Adafruit_SHT31.cpp#L186). 
This could probably be blocking further code execution since it basically brings the µC to a halt. You could try to modify that and see if it helps in any way.

closed 2019-03-04 by Avamander