nRF24/RF24Mesh #156: battery powered sensor network using rf24mesh

Zuputmer, 2018-12-08:
As the title suggests I am trying to implement a mesh network that consists of nodes which are battery powered, and I am using the 328p as a standalone microcontroller without the Arduino hardware so as to reduce power consumption, and just by flashing the Arduino bootloader to it. My question is, is the mesh network capable of relaying a message by waking up a node just to relay a message to another node? Should I use the IRQ as an external interrupt to wake up the 328p? If so, what command should I run in the interrupt service routine of the 328p?

TMRh20, 2018-12-09:
Yes,The AVR device can sleep but the radios need to remain active (15ma)

> Should I use the IRQ as an external interrupt to wake up the 328p?

Sure.

Zuputmer, 2018-12-10:
@TMRh20 Thank you for your reply. Unfortunately I cannot keep the radios active all the time as it would severely affect the life of my modules. Is there any other way? What if I used the standby 1 or 2 modes as suggested in the datasheet? 
If some of my nodes needed to only transmit and not relay any information, could I turn them off completely for a specific period once I obtained the acknowledge from the receiver?
When a relaying node has received a packet that it needs to send to another node, will it still give IRQ signal?

Thanks in advance.

Zuputmer, 2018-12-10:
@ristomatti thank you for your reply. Does the relaying node need any special code? Or is it just a normal node running the same code that needs to be always on? Sorry if my question sounds stupid, I am new to this :)

closed 2018-12-14 by Zuputmer