nRF24/RF24Mesh #157: Mesh addressing

Zuputmer, 2018-12-27:
Hey,
I was wondering, supposing i have 100 nodes running the mesh sketch and one master in the middle. Supposing all 100 nodes are within the range of the master, how will the nodes be addressed then? Will the tree topology still be followed or will they directly talk to the master? What will their addresses look like?

Please forgive my question, i am new to this. Thanks in advance.

TMRh20, 2018-12-27:
The tree topology and routing will still be followed just the same
regardless of number of nodes or proximity to master.

Zuputmer, 2018-12-27:
@TMRh20 Thank you for your reply. However, in my network all nodes will not be powered on at any given time. The nodes will be powered on one after the other and then powered off after they finish sending data to the master,, based on an interrupt from an RTC. If routing is still followed regardless of proximity to the master, then i cannot implement a mesh network using my individual battery powered nodes, am i right?

Zuputmer, 2018-12-28:
@TMRh20 Thank you so much for your reply. Once again, your help is greatly appreciated. I now have a clear idea of what i should do. However, i do have a couple of questions-
1)Since auto retry will not happen as you mentioned, i can use the returned bool variable from the write function to check if transmission has failed and then retry manually in the code, am i right?
2)When using unicast write, is it possible for the master to know the node ID of the node that sent the data to it? Or should i include the address in the payload being sent to the master?

TMRh20, 2018-12-28:
No problem.

1. When using unicast, there is no acknowledgement, so no way to know if a payload was successful.

2. The master node will have the network address of the node that sent the payload, which can be translated to node ID by the master if using RF24Mesh.

closed 2019-01-03 by Zuputmer