nRF24/RF24Mesh #158: RF24MESH NOT WORKING

ajinkyaw, 2019-01-28:
I have developed a program for Master and Slave devices using NRF24L01 Devices.
I am using RF24MESH Library to achieve my requirements. 
Till yesterday everything was working but all of a sudden from morning it is not working.
I tried to use the RF24MESH_MESH_EXAMPLE_MASTER AND RF24MESH_EXAMPLE , which i was used to TEST if NRFs are working or not. This example is also not working now.
The Slave is not be able to connect to the Master.
I tried to run ping-pair example, It is working fine.

Please  guide me to solve this issue.
Note ; I have connected 470uF cap on Nrf's terminals.

Avamander, 2019-01-28:
> This example is also not working now.

Check the connections to the modules.

ajinkyaw, 2019-01-28:
**Device 1**

RF24/examples/pingpair_ack/

ROLE: Pong back
*** PRESS 'T' to begin transmitting to the other node
STATUS		 = 0x0e RX_DR=0 TX_DS=0 MAX_RT=0 RX_P_NO=7 TX_FULL=0
RX_ADDR_P0-1	 = 0x544d52687c 0xabcdabcd71
RX_ADDR_P2-5	 = 0x33 0xce 0x3e 0xe3
TX_ADDR		 = 0x544d52687c
RX_PW_P0-6	 = 0x01 0x01 0x20 0x20 0x20 0x20
EN_AA		 = 0x3f
EN_RXADDR	 = 0x3e
RF_CH		 = 0x4c
RF_SETUP	 = 0x07
CONFIG		 = 0x0f
DYNPD/FEATURE	 = 0x03 0x06
Data Rate	 = 1MBPS
Model		 = nRF24L01+
CRC Length	 = 16 bits
PA Power	 = PA_MAX

**Device 2**

RF24/examples/pingpair_ack/

ROLE: Pong back
*** PRESS 'T' to begin transmitting to the other node
STATUS		 = 0x0e RX_DR=0 TX_DS=0 MAX_RT=0 RX_P_NO=7 TX_FULL=0
RX_ADDR_P0-1	 = 0x544d52687c 0xabcdabcd71
RX_ADDR_P2-5	 = 0x33 0xce 0x3e 0xe3
TX_ADDR		 = 0x544d52687c
RX_PW_P0-6	 = 0x01 0x01 0x20 0x20 0x20 0x20
EN_AA		 = 0x3f
EN_RXADDR	 = 0x3e
RF_CH		 = 0x4c
RF_SETUP	 = 0x07
CONFIG		 = 0x0f
DYNPD/FEATURE	 = 0x03 0x06
Data Rate	 = 1MBPS
Model		 = nRF24L01+
CRC Length	 = 16 bits
PA Power	 = PA_MAX
*** CHANGING TO TRANSMIT ROLE -- PRESS 'R' TO SWITCH BACK
Now sending 1 as payload. Got response 1, round-trip delay: 7084 microseconds

Now sending 2 as payload. Got response 2, round-trip delay: 1528 microseconds

Now sending 3 as payload. Got response 3, round-trip delay: 584 microseconds

Now sending 4 as payload. Got response 4, round-trip delay: 1048 microseconds

Now sending 5 as payload. Got response 5, round-trip delay: 2920 microseconds

Now sending 6 as payload. failed.
Now sending 6 as payload. Got response 6, round-trip delay: 588 microseconds

Now sending 7 as payload. Got response 7, round-trip delay: 1972 microseconds

Now sending 8 as payload. Got response 8, round-trip delay: 2452 microseconds

Now sending 9 as payload. Got response 9, round-trip delay: 1048 microseconds

Now sending 10 as payload. Got response 10, round-trip delay: 1048 microseconds

Now sending 11 as payload. Got response 11, round-trip delay: 584 microseconds

Now sending 12 as payload. Got response 12, round-trip delay: 584 microseconds

Now sending 13 as payload. Got response 13, round-trip delay: 1048 microseconds

Now sending 14 as payload. Got response 14, round-trip delay: 2448 microseconds

Now sending 15 as payload. Got response 15, round-trip delay: 4760 microseconds

Now sending 16 as payload. Got response 16, round-trip delay: 584 microseconds

Now sending 17 as payload. Got response 17, round-trip delay: 1080 microseconds

Now sending 18 as payload. Got response 18, round-trip delay: 1048 microseconds

Now sending 19 as payload. Got response 19, round-trip delay: 2452 microseconds

Now sending 20 as payload. Got response 20, round-trip delay: 1508 microseconds

Now sending 21 as payload. Got response 21, round-trip delay: 1044 microseconds

Now sending 22 as payload. Got response 22, round-trip delay: 580 microseconds

Now sending 23 as payload. Got response 23, round-trip delay: 576 microseconds

Now sending 24 as payload. Got response 24, round-trip delay: 1044 microseconds

Now sending 25 as payload. Got response 25, round-trip delay: 7088 microseconds

Now sending 26 as payload. Got response 26, round-trip delay: 2916 microseconds

Now sending 27 as payload. Got response 27, round-trip delay: 1528 microseconds

Now sending 28 as payload. Got response 28, round-trip delay: 576 microseconds

Now sending 29 as payload. Got response 29, round-trip delay: 1532 microseconds

Now sending 30 as payload. Got response 30, round-trip delay: 1508 microseconds

Avamander, 2019-01-28:
The variance in the round-trip delay and the high lost packet rate signifies that there is some problem in the system. Do try to increase decoupling to 10 000uF (and a small ceramic capacitor to counter the big inductance of the electrolytic capacitor). You can also try and increase the power a bit.

ajinkyaw, 2019-01-28:
I have developed my own pcb . On NRF terminals i have connected 470uF cylinder capacitor and 104 pF capacitor too.
I will try to increase to 10,000uF but after doing a lot of research on internet i have found this pair of 470uF and 104pF to connect to terminals. This system was working fine since last 7 days , today morning only something has happened.And i am not be able to find it.

Avamander, 2019-03-04:
>  today morning only something has happened.And i am not be able to find it.

Yeah then it's probably some environmental change. There's nothing I/this library can do against that.

closed 2019-03-04 by Avamander