nRF24/RF24Mesh #160: Doubt on mesh working

Zuputmer, 2019-02-11:
Hey, i have a couple of questions and i would be grateful if you could answer them.

Assume that I have a mesh with 100 slave nodes. Initially, all the nodes are turned completely off ( AVR in PWR_DOWN and nrf turned off with a transistor ). The nodes are set to wake in predetermined intervals of time via the INT0 ext interrupt through an RTC. If all the nodes are woken up simultaneously and send data at the same instant of time, what will happen? Will there be a collision? Will the network automatically resolve the collisions? WIll the master be able to receive the data sent by all the nodes at the same instant?

If not, i also have the option to wake up nodes in any order with any amount of time gap between them. For example if i wake up the second node 100ms after the first node, will that help? To what extent will i be able to reduce this time gap, since i am looking to reduce the ON time of the slave nodes because power consumption is critical?

TMRh20, 2019-02-14:
*"If all the nodes are woken up simultaneously and send data at the same
instant of time, what will happen?"*

Assuming you are using RF24Network, and all nodes wake up at the same time
and send data at the same instant, the data will collide, and nothing much
will happen.

*"For example if i wake up the second node 100ms after the first node, will
that help? "*

Yes.

*"To what extent will i be able to reduce this time gap, since i am looking
to reduce the ON time of the slave nodes because power consumption is
critical?  "*

 You would have to experiment to find the smallest time gap, I am not sure.

<http://www.avg.com/email-signature?utm_medium=email&utm_source=link&utm_campaign=sig-email&utm_content=webmail>
Virus-free.
www.avg.com
<http://www.avg.com/email-signature?utm_medium=email&utm_source=link&utm_campaign=sig-email&utm_content=webmail>
<#DAB4FAD8-2DD7-40BB-A1B8-4E2AA1F9FDF2>

Zuputmer, 2019-02-15:
@TMRh20 thanks for your reply. 

> Assuming you are using RF24Network, and all nodes wake up at the same time and send data at the same instant, the data will collide, and nothing much will happen.

Actually my doubt was what would happen if i used RF24Mesh and the slaves woke up simultaneously and transmitted at the same instant of time.

TMRh20, 2019-02-15:
It kind of depends how you have things set up and working, but typically,
nodes will re-transmit when a collision occurs, but if 100 nodes are
sending, the chances of data making it through is limited.

If nodes need to renew addresses etc., it complicates things further. If
you are using a unicast write, direct to the master, things will still work
differently, and nodes need to stagger their transmissions.

All in all, I would say you need to experiment with your setup a bit to see
what works and what doesn't.

<http://www.avg.com/email-signature?utm_medium=email&utm_source=link&utm_campaign=sig-email&utm_content=webmail>
Virus-free.
www.avg.com
<http://www.avg.com/email-signature?utm_medium=email&utm_source=link&utm_campaign=sig-email&utm_content=webmail>
<#DAB4FAD8-2DD7-40BB-A1B8-4E2AA1F9FDF2>

closed 2019-02-18 by Zuputmer