nRF24/RF24Mesh #163: I can't communicate a node with another node.

wavewolf, 2019-04-05:
Communication from any node to master works correctly but if I want to send master data to a specific node it does not work. It also happens when I want to send data from one node to another node, it does not work.

TMRh20, 2019-04-06:
Not sure what to suggest. If the mesh is up and functioning it should just work.

wavewolf, 2019-04-06:
I explain the situation. I have 4 systems with nrf24l01 + modules:

a- System 1 Master (receives information and sends information to the nodes).
b- Node 1 system (sends information to the master and node 3)
c- Node 2 system (sends information to the master and node 3)
d - System node 3 (receives infomation to the node 1 & node 2 and sends information to the master)

I only get it to work when the nodes send information to the master, any other option doesn't work. I can't send information from node to node or from master to node.

TMRh20, 2019-04-07:
Only thing I can think of is to try a different radio channel in case of interference. Use mesh.begin(uint8_t channel)

wavewolf, 2019-04-07:
I don't know if it has any sense with it, but when I use the normal RF24 library and I define reading channels in the supposed master, it only works in this way: openReadingPipe(1, pipes[1]), openReadingPipe(1, pipes[2]), openReadingPipe (1, pipes[3])... but if I try: openReadingPipe(1, pipes[1]), openReadingPipe(2, pipes[2]), openReadingPipe (3, pipes[3]), it doesn't work. The master don't receive anything. This problem could be the same?

TMRh20, 2019-04-09:
Yeah, that seems like the same problem. Really odd.

Avamander, 2019-04-17:
What's your setup like? How well decoupled is the power? Are you sure your modules are not counterfeit? What Arduinos are you using?

closed 2020-08-16 by TMRh20