nRF24/RF24Mesh #165: Master unable to send to node, if AddressList list has been cleared

systemofapwne, 2019-06-30:
Hello,
whenever the AddressList has been cleared, e.g. when the master restarts (without saving in dhcplist.txt) but client still has a valid address, the client might be able to still send data to the master and the master receives it. Yet, the master is unable to send data back to the client, because it's AddressList is empty: The network header resolves to "from_nodeID = -1".

I have to add, that I use this nodejs wrapper of your lib (https://github.com/ludiazv/node-nrf24). But tracking down the error, it all went into the direction of RF24Mesh library.

**Suggested fix**: When the master receives from a node he has not yet in his AddressList, add him to the AddressList or _force_ him to reaquire a new address (e.g. by causing _RF24Mesh::checkConnection()_ to fail on the remote node)
By the way: Checking connectivity via _RF24Mesh::checkConnection()_ on the node will return true in such situations. So the node still thinks, he is properly connected, while it isn't (because just in one direction).

Avamander, 2019-06-30:
This has been discussed a few times. The general consensus is that everyone has to implement a method of checking if you have a valid address, all misconfigurations are too complex to detect and fix in the space available on an Arduino. Address validity check can be done by implementing a trivial ping-pong system you can also use for other purposes, like updating time on leaf nodes.

systemofapwne, 2019-06-30:
I was not aware of this situation. In the mean-time I implemented a ping-pong system as a workaround, as you also have suggested.
I just thought, adding a quick fix to the library itself, wouldn't hurt but I agree on the complexity.

Thank you for your reply!

closed 2019-06-30 by systemofapwne