nRF24/RF24Mesh #167: RF24mesh  setNodeID

systeams, 2019-07-18:
Hello everyone 

i'm working on a little project where i have to connect some arduino UNO using RF24Mesh to a raspberryPI 3 B+ 

i used the example of the github nRF24/RF24Mesh on both sides the issus that i'm facing now is when i use 

> mesh.setNodeID(3)   // NodeID (Node Side) as you know master is set to 0 

the node is unable to connect to the network and stuck at Renewing Address 
when i remove it the node connect to the network but no address is attributed by the Master and i get data received "SOME DATA" from 00
and when i print out the address list it is empty 
what can cause this problem 

thank you in advance for your help

tomcatGu, 2019-07-18:
I think the reason is that the timeout value is too small at line 211 and line 355 in RF24Mesh.cpp.
```
line 211 :
uint32_t timer=millis(), timeout = 125;

line 355:
while(millis()-timr<225)
```

systeams, 2019-07-18:
@tomcatGu  
i checked the file my timeout is 500 in line 211 and the line 355 is still ( <225) the same , what i don't understand when i use the Master example on an Arduino and Node example on an other one it works perfectly but as soon as i use the master example on a raspberry it doesn't work until i remove the setNodeID on the node side

tomcatGu, 2019-07-18:
Master node ID must set to 00.

tomcatGu, 2019-07-18:
@systeams 
i test your method.That work fine when there is only one client node.
if there have more than two client nodes the master node prints the following error:
`MSH: Invalid id 0 rcvd`
The mesh network is very instable.

systeams, 2019-07-20:
@tomcatGu 
finally i fixed the problem i changed the delay of the while loop to 1 and and every think work fine now i also shut down the raspberrypi before trying again and the network is working just fine with 4 nodes 
@tomcatGu  thank you for your orientation about the timeout

Ademates22, 2020-03-08:
@systeams hey, can you please tell which while loop delay you have changed? is it a while in mesh.cpp?

closed 2020-08-16 by TMRh20