nRF24/RF24Mesh #168: Temporarily Unlinking a node

itisyeetimetoday, 2019-10-16:
Hi,
Is it possible to use this library for transmitting telemetry data, then using this library to send out a packet of information, like telling the module to become a repeater? After that, is it still possible to have the node rejoin the network?

TMRh20, 2019-10-16:
Umm, yes?

TMRh20, 2019-10-16:
> Do you have an example about how to have 2 master nodes and how to send a message from the master node to the a slave node to display like on lcd?

No example, but the only way to have 2 separate master nodes is to start them up on separate channels. Then the node(s) can just switch channels to switch between masters.

itisyeetimetoday, 2019-10-16:
Thank you! However, is there a way to send a message from the master to the slave?

closed 2020-08-16 by TMRh20