nRF24/RF24Mesh #169: network adress

eldiamond2, 2019-11-17:
i have gateway and nodes working thru RF24MESH.h

some of gateway code:
```
void handleNRF24Network() {
  RF24NetworkHeader answer;
  if(network.available()){
  switch (header.type){
    -=-= message handling code there =-=-
  case ID_RESOLVE: // system header for UniqueID get
  // my solution to get unique ID for nodes
      network.read(header,&_Value._8_t_param, sizeof(_Value._8_t_param));
      if(_Value._8_t_param == 'A') {
        Serial.print("\nUnique ID request accepted: ");
        newID = EEPROM.read(0x0000); // read
        if(newID < 3 || newID == 255) {newID = 3;} // reset id's
        Serial.printf("[0x0000]:%d", newID);
        Serial.printf("\n To %d will be [%d]\n", NodeID, newID);
        answer.from_node = 0;
        answer.id = header.next_id;
        answer.to_node = header.from_node;
        answer.type = ID_RESOLVE;
        network.write(answer, &newID, sizeof(newID));
        newID++;
        Serial.print("Sending to: "); Serial.print(header.from_node);
        EEPROM.write(0x0000, newID); // save
  // nodes then remember this ID and don't ask it after, it's becomes unique for current node on it's side when it's connecting to mesh
  }
    break;
  }
}

void loop() {
  mesh.update();  
  mesh.DHCP();
  mqtt.loop();
  handleNRF24Network();
  // -=-= some other functional code =-=-
}

```

but when gateway restarting, and node - not, and i try to resolve:
`mesh.getAddress(node->getID()`
fuction returns -1, or, to be exactly 65535, and message data don't get to node
is there any possibility to send data by ID, or some any way? how to send data when adress not resolved?

TMRh20, 2019-11-17:
Addresses pretty much need to be resolvable in order to send from master to children. Two options, 1. You could leave the master offline long enough for children to detect it snd renew addresses or 2. Have the nodes do an address lookup on their own nodeid every 30 seconds or so and renew addresses if it fails. This should ensure nodes are almost always available.

eldiamond2, 2019-11-18:
> Addresses pretty much need to be resolvable in order to send from master to children. Two options, 1. You could leave the master offline long enough for children to detect it snd renew addresses or [...]
first opinion is not an option. 
Is there any build in function to node address look up? 
there are renewAddress() function, calling it once in 2-5 minutes should be enough?

TMRh20, 2019-11-18:
Yeah, mesh.getAddress() See https://tmrh20.github.io/RF24Mesh/classRF24Mesh.html#a895f444dcc0cb175faf845c5bdd7a7d0

Alternately yes, you could just renew the addresses periodically.

eldiamond2, 2019-11-18:
> Yeah, mesh.getAddress() See https://tmrh20.github.io/RF24Mesh/classRF24Mesh.html#a895f444dcc0cb175faf845c5bdd7a7d0 Alternately yes, you could just renew the addresses periodically.
> […](#)

Thanks!
With best regards!

closed 2019-11-18 by eldiamond2