nRF24/RF24Mesh #170: Mesh is not connecting more than 5 modes

ksnboopa, 2020-01-04:
Hi,

I have PI zero as master and arduino nano as my sensor nodes. When I connect to my nodes everything seems to works fine with samples. When I add one mode node to mesh, mesh is connected but not able to send messages to all my nodes.

Master : RPi Zero
Nodes  : Arduino nano boards

I will change only node ID to add any new node and upload my sketch.  

Whenever I get MQTT msg, will be published to specific node. 
Whenever I get Mesh msg, will be published to MQTT back. 

I tried all possible ways for this. Please help me on this. Or should I enable any configuration If I want to add multiple nodes while I use fixed nodeID.

Child script : 

int FREQ = 98; // Subject to change
uint8_t thisNode = 5; // Unique value for each node
String moduleType = "2SWMDL";

const int outPinsLength = 2;
int eepromStartIndex = 11;
int outPins[] = { 3, 5 };
int meshUpdateInterval = 0;
RF24 radio(7, 8);
RF24Network network(radio);
RF24Mesh mesh(radio, network);

#define STATUS_LED_PIN 2
#define DATA_LED_PIN 6

int pinStatues[outPinsLength];

uint32_t displayTimer = 0;

bool powerLossHandled = false;

struct request {                 // Structure of our payload
	uint8_t nodeId;
	char action[15];
	char data[25];
};

struct response {                 // Structure of our payload
	uint8_t nodeId;
	char module[15];
	char data[25];
};

void setup() {
	Serial.begin(115200);
	pinMode(STATUS_LED_PIN, OUTPUT);
	pinMode(DATA_LED_PIN, OUTPUT);
	blink(STATUS_LED_PIN, 4);
	blink(DATA_LED_PIN, 4);
	MsTimer2::set(10, powerLossDetector);
	MsTimer2::start();
	EEPROM.begin();
	int localEEpromIndex = eepromStartIndex;
	for (int i = 0; i < outPinsLength; i++) {
		uint8_t state = EEPROM.read(localEEpromIndex);
		pinStatues[i] = state ? HIGH : LOW;
		pinMode(outPins[i], OUTPUT);
		digitalWrite(outPins[i], pinStatues[i]);
		localEEpromIndex += 1;
	}
	// Set the nodeID manually
	mesh.setNodeID(thisNode);
	// Connect to the mesh
	Serial.println(F("Connecting to the mesh..."));
[... 379 lines cut ...]
		while (network.available()) {
			RF24NetworkHeader header;
			network.peek(header);

			response dat;
			switch (header.type) {
			case 'M': {
				network.read(header, &dat, sizeof(response));
				cout << "<< GOT FROM NODE : " << (int) dat.nodeId << ", MODULE: "
						<< dat.module << ", DATA: " << dat.data << endl;
				process_node_response(dat);
				break;
			}
			default: {
				network.read(header, 0, 0);
				printf("Rcv bad type %d from 0%o\n", header.type,
						header.from_node);
				break;
			}
			}
		}
		delay(2);
	}
	return 0;
}

Avamander, 2020-01-04:
You aren't calling `mesh.update()` often enough for any reliable node-to-node communication required for the mesh to function.

ksnboopa, 2020-01-05:
@Avamandar,

I am calling it in master even without delay and in child node in every 4 seconds. 

Could that be a problem?

I’ll change it without delay in child and let me check.

ksnboopa, 2020-01-05:
No luck yet.

Avamander, 2020-01-05:
If you take the bare examples, do they work on your hardware and with this node count?

ksnboopa, 2020-01-07:
With bare examples I have not tested it yet. We will change implementation to RF24Ethernet and check it and let know

ksnboopa, 2020-01-09:
@Avamander,

It was memory issue. my SRAM was used more than 73%. Due to which program was not stable enough as atmel 328 has 2048bytes only. Already Ethernet along with rf24mesh took around 68% progmem and 65% SRAM for mqtt_basic.ini itself. 
 
I am just monitoring my devices and it’s stability with 8 nodes along with Raspberry gateway. I will post results after a week monitoring.

Qn. I am planning to connect 100 nodes with a single Raspberry gateway with data rate as 1 mbps. Will it be a good enough design or do I need to include any extra care with code and config.?

Anyways thanks a lot for your support and this awesome library guys...

Avamander, 2020-01-09:
> It was memory issue. my SRAM was used more than 73%. 

Wrap every string in your code into `F()` for example `F("This is a string")` that will reduce RAM usage.

> Will it be a good enough design or do I need to include any extra care with code and config.?

The library can not handle collisions, either do scheduling or retransmissions if you need arrival.

>  I am planning to connect 100 nodes with a single Raspberry gateway with data rate as 1 mbps.

If you reset your master node then you either have to load address allocations from disk and/or implement a ping-pong scheme to make sure nothing is lost. Have your nodes contact the master and expect a reply, that way the node will know the address it has is valid, if the address is not valid but your node uses it it'll cause issues.

closed 2020-08-16 by TMRh20