nRF24/RF24Mesh #171: Building a miltiple devices(200+) long-range network.

wow4er, 2020-01-31:
Hello, i'm planning to use NRF24 modules without amplifier. 
And quesiton is how is exactly this MESHnetwork works?
For example i have:
- Master Node.
- SensorsNode1 50 meters far from the master.
- SensorsNode2 100 meters far from the master.

Master is receive data from the SensorsNode1, but get nothing from the 100 meters distance.
I thought that in this case SensorsNode2 should to find a way to send data to the master via SensorsNode1? Or i'm wrong? 
I'm planning to add more sensors on distance 200 and 300 meters, using same NRF modules, but stuck on this issue.
Both modules working fine from 50 meters distance.

Thank you in advance for you answers.
Best Regards,
Vladimir.

TMRh20, 2020-01-31:
It’s hard to say what the problem is, but your basic assumptions about how it works seem to be correct. 

Traffic from node2 should get routed through node1 to get to the master.

If your nodes will be stationary, you may have better luck using RF24Network instead of the mesh overlay.

Else with RF24Mesh you can enable debugging on node1 by editing the config files to see if it is receiving and routing packets from node2.

wow4er, 2020-02-03:
Main idea is make this network dynamic, modules wont be a same place and will be moved, that's why i choosen  Mesh solution, but i'm not sure how it supposed to work. If it wont be hard for you, can you take a quick look at the sketch to see whats i'm doing wrong. Because this way i see that modules working only in Sensor -> master mode. In case of long range and no connection for a "Sensor2",  this not working like Sensor2 -> Sensor1 -> Master.
Thank you for a quick response.
[sensor-mesh.zip](https://github.com/nRF24/RF24Mesh/files/4147707/sensor-mesh.zip)
[master-mesh.zip](https://github.com/nRF24/RF24Mesh/files/4147713/master-mesh.zip)

Avamander, 2020-02-03:
Enable debugging and see in your own setup if and how the nodes communicate.

In order to find out if it's your code or the physical hardware, upload the bundled examples to the boards and see if those work.

TMRh20, 2020-02-03:
One thing I notice in your code is the setting to 250kbps speed. The mesh was tested etc at 1mbps, so I would also test at 1mbps data rate just in case the delays etc are too small for 250kbps.

closed 2020-08-16 by TMRh20