nRF24/RF24Mesh #172: RF24Mesh Library on the Arduino Due

bnmangirazi, 2020-04-05:
Has anyone managed to use the library with the Arduino Due successfully without making major changes to the library?

Avamander, 2020-04-05:
What's stopping you now?

bnmangirazi, 2020-04-05:
I have a mesh network of Arduino Uno's. Now i want to add another node of the Arduino Due, it seems to be connecting properly to the network but the output from the master node is wrong, its not the same as the other nodes in the network yet they are all running similar codes, except for the physical SPI connection with the NRF24L01+ modules.

Avamander, 2020-04-05:
If the output is wrong you might've forgotten to take into account data type size changes between the platforms.

bnmangirazi, 2020-04-05:
Okay. i surely did not change anything on the code except for the node ID's and SPI config.

bnmangirazi, 2020-04-05:
Let me try to work around it and give feed back.

bnmangirazi, 2020-04-05:
Any heds

> If the output is wrong you might've forgotten to take into account data type size changes between the platforms.

I would appreciate any heads up

bnmangirazi, 2020-04-06:
So it turns out i was wrong about the connection to the Mesh Network. the Due is not even connecting, i have tried other examples including the RF24 helloworld but it's not working. on one end i have an arduino uno hooked up to an RF24L01+ which i am sure is working and on the remote end i have an arduino Due hooked up to an RF24L01+. For the Due I am using the SPI pins located near the centre of the board, MISO-MISO(pin1), SCK-SCK(pin3), MOSI-MOSI(pin4). VCC is connected to 3.3V, GND to GND.

I need to at least get the helloworld example working between the UNO and the Due to get me going.

Please assist, i am stuck.#Newbie

bnmangirazi, 2020-04-06:
> So it turns out i was wrong about the connection to the Mesh Network. the Due is not even connecting, i have tried other examples including the RF24 helloworld but it's not working. on one end i [...]

CE is pin 8, CSN is pin 9.

**helloworldtx**
#include <RF24Network.h>
#include <RF24.h>
#include <SPI.h>

RF24 radio(8,9);                    // nRF24L01(+) radio attached using Getting Started board 

RF24Network network(radio);          // Network uses that radio

const uint16_t this_node = 01;        // Address of our node in Octal format
const uint16_t other_node = 00;       // Address of the other node in Octal format

const unsigned long interval = 2000; //ms  // How often to send 'hello world to the other unit

unsigned long last_sent;             // When did we last send?
unsigned long packets_sent;          // How many have we sent already

struct payload_t {                  // Structure of our payload
  unsigned long ms;
  unsigned long counter;
};

void setup(void)
{
  Serial.begin(115200);
  Serial.println("RF24Network/examples/helloworld_tx/");
 
  SPI.begin(10);
  SPI.begin(4);
  radio.begin();
  network.begin(/*channel*/ 90, /*node address*/ this_node);
}

void loop() {
  
  network.update();                          // Check the network regularly

  
  unsigned long now = millis();              // If it's time to send a message, send it!
  if ( now - last_sent >= interval  )
  {
    last_sent = now;

    Serial.print("Sending...");
    payload_t payload = { millis(), packets_sent++ };
    RF24NetworkHeader header(/*to node*/ other_node);
    bool ok = network.write(header,&payload,sizeof(payload));
    if (ok)
      Serial.println("ok.");
    else
      Serial.println("failed.");
  }
}

**helloworldrx**

#include <RF24Network.h>
#include <RF24.h>
#include <SPI.h>

RF24 radio(8,9);                // nRF24L01(+) radio attached using Getting Started board 

RF24Network network(radio);      // Network uses that radio
const uint16_t this_node = 00;    // Address of our node in Octal format ( 04,031, etc)
const uint16_t other_node = 01;   // Address of the other node in Octal format

struct payload_t {                 // Structure of our payload
  unsigned long ms;
  unsigned long counter;
};

void setup(void)
{
  Serial.begin(115200);
  Serial.println("RF24Network/examples/helloworld_rx/");
 
  SPI.begin();
  radio.begin();
  network.begin(/*channel*/ 90, /*node address*/ this_node);
}

void loop(void){
  
  network.update();                  // Check the network regularly

  
  while ( network.available() ) {     // Is there anything ready for us?
    
    RF24NetworkHeader header;        // If so, grab it and print it out
    payload_t payload;
    network.read(header,&payload,sizeof(payload));
    Serial.print("Received packet #");
    Serial.print(payload.counter);
    Serial.print(" at ");
    Serial.println(payload.ms);
  }
}

TMRh20, 2020-04-06:
Sounds like you are correctly using the ICSP header, but I would still
suggest double checking your pin configuration here:
https://www.arduino.cc/en/reference/SPI

 The wiring/configuration are going to be the most likely reasons why it
isn't working for you.

Make sure you connect the radio to the 3.3v pin, and NOT the ICSP header
vcc which is 5v.

Also you don't need to call SPI.begin().

<http://www.avg.com/email-signature?utm_medium=email&utm_source=link&utm_campaign=sig-email&utm_content=webmail>
Virus-free.
www.avg.com
<http://www.avg.com/email-signature?utm_medium=email&utm_source=link&utm_campaign=sig-email&utm_content=webmail>
<#DAB4FAD8-2DD7-40BB-A1B8-4E2AA1F9FDF2>

<http://www.avg.com/email-signature?utm_medium=email&utm_source=link&utm_campaign=sig-email&utm_content=webmail>
Virus-free.
www.avg.com
<http://www.avg.com/email-signature?utm_medium=email&utm_source=link&utm_campaign=sig-email&utm_content=webmail>
<#DAB4FAD8-2DD7-40BB-A1B8-4E2AA1F9FDF2>

bnmangirazi, 2020-04-06:
I have verified the connections and power to the radio module,  but still not working.
I have done a simple test to check if the SPI is working on the Due using some example i found on the internet. 

#define TEST_MOSI 74
#define TEST_MISO 75
#define TEST_CLK 76

void setup() {
  pinMode(TEST_MOSI, OUTPUT);
  pinMode(TEST_MISO, OUTPUT);
  pinMode(TEST_CLK, OUTPUT);
}

void loop() {
  delay(1000);
  digitalWrite(TEST_MOSI, 0);
  digitalWrite(TEST_CLK, 0);
  digitalWrite(TEST_MISO, 0);
  delay(1000);
  digitalWrite(TEST_MOSI, 1);
  digitalWrite(TEST_CLK, 1);
  digitalWrite(TEST_MISO, 1);
}

The voltage is flactuating between 0v and 3.28v on all the pins (MISO, SCK & MOSI).

TMRh20, 2020-04-06:
That's not really a test for SPI, only for GPIO control.

To see if you have SPI communication with the radio module and Due call `radio.printDetails();` after calling network.begin(). 

Then if you unplug/reset the Due, it should output the following:
```
RF24Network/examples/helloworld_tx/
STATUS		 = 0x0e RX_DR=0 TX_DS=0 MAX_RT=0 RX_P_NO=7 TX_FULL=0
RX_ADDR_P0-1	 = 0xcccccc3ccc 0xcccccc3c3c
RX_ADDR_P2-5	 = 0x33 0xce 0x3e 0xe3
TX_ADDR		 = 0xcccccccc3c
RX_PW_P0-6	 = 0x20 0x20 0x20 0x20 0x20 0x20
EN_AA		 = 0x3e
EN_RXADDR	 = 0x3f
RF_CH		 = 0x5a
RF_SETUP	 = 0x07
CONFIG		 = 0x0f
DYNPD/FEATURE	 = 0x3f 0x04
Data Rate	 = 1MBPS
Model		 = nRF24L01+
CRC Length	 = 16 bits
PA Power	 = PA_MAX
```
If these settings do not match what you see, you have an SPI communication problem, again most likely wiring/configuration related or even a hardware failure.
If they do match, everything should work as desired, try unplugging the devices to ensure default radio settings etc. and re-test.

bnmangirazi, 2020-04-06:
My output is different.

RF24Network/examples/helloworld_tx/
STATUS		 = 0x00 RX_DR=0 TX_DS=0 MAX_RT=0 RX_P_NO=0 TX_FULL=0
RX_ADDR_P0-1	 = 0x0000000000 0x0000000000
RX_ADDR_P2-5	 = 0x00 0x00 0x00 0x00
TX_ADDR		 = 0x0000000000
RX_PW_P0-6	 = 0x00 0x00 0x00 0x00 0x00 0x00
EN_AA		 = 0x00
EN_RXADDR	 = 0x00
RF_CH		 = 0x00
RF_SETUP	 = 0x00
CONFIG		 = 0x00
DYNPD/FEATURE	 = 0x00 0x00
Data Rate	 = 1MBPS
Model		 = nRF24L01
CRC Length	 = Disabled
PA Power	 = PA_MIN

I have cross checked the connection quite a considerable number of times now, I am convinced i have a hardware problem on the SPI interface.

TMRh20, 2020-04-07:
Unfortunate if that is the case.
[This link](https://forum.arduino.cc/index.php?action=dlattach;topic=479176.0;attach=212948) shows the correct layout for the SPI ICSP pins, just in case lol.

There are not many other options on the Due such as softSPI. There is a possibility to use the Due USARTs for SPI, but I haven't adapted my library for it, and I don't think anybody else has made one either.

TMRh20, 2020-04-07:
Err, in looking things over for something else I'm working on, I realized that my UART_SPI driver actually does support the Due. I must have added support and forgot about it.

In any case, you can actually prove this out if you would like to try it out:
1. Download [SPI_UART library (in sketch folder) ](https://github.com/TMRh20/Sketches/archive/master.zip)
2. Extract the SPI_UART folder to your arduino library folder: `Documents/Arduino/libraries/SPI_UART`
3. Edit RF24_config.h : uncomment `#define SPI_UART`
4. Reconnect your rf24 module: `MOSI->TX1 MISO->RX1 SCK->SDA1 `   CS&CE the same
   (SDA1 is down past pin 13)

It should work exactly the same, no other changes needed.
If this works but the ICSP pins don't, I think you can confirm a hardware issue there, but at least you would still have functioning SPI.

bnmangirazi, 2020-04-07:
i have tried the above and somehow, my program for helloworldtx cannot go past the setup phase.

TMRh20, 2020-04-07:
[This line](https://github.com/TMRh20/Sketches/blob/master/SPI_UART/SPI_UART.cpp#L160)
can be changed to work around the issue.
`USART0->US_BRGR = 9;`

It seems something changed in the RF24 library, haven't used the usart lib for a while, but that should correct it.  Sorry bout that.

bnmangirazi, 2020-04-07:
Thank you so much, the program is now running beyond  "setup", however i'm getting the same results as before with the ICSP SPI.

TMRh20, 2020-04-07:
Have you tried a different radio with the due? One that works for sure on another device?

bnmangirazi, 2020-04-07:
Yes i have i am sure the radio is fine. i even tested it on the Uno. i also tried swapping the radios. i am left to believe there could be something wrong with my Due.

TMRh20, 2020-04-07:
Ok, I think you have done about as much troubleshooting as can be expected. To go any further it would probably be best to have like an oscilloscope or logic analyzer to see wtf is going on with those SPI pins... if you do manage to figure something out, please let us know the cause of this.

Avamander, 2020-04-07:
One more thing you can try is checking if you're on the latest version of both Arduino boards support and RF24*.

bnmangirazi, 2020-04-07:
> Ok, I think you have done about as much troubleshooting as can be expected. To go any further it would probably be best to have like an oscilloscope or logic analyzer to see wtf is going on with those SPI pins... if you do manage to figure something out, please let us know the cause of this.

Thanks, i will check with an oscilloscope after the lock down. meanwhile i have ordered another Due from a different supplier.
 
i will give updates in due course.

closed 2020-08-16 by TMRh20