nRF24/RF24Mesh #173: RF24Mesh library doesn't install correctly in PlatformIO

microtronics, 2020-05-17:
When adding the RF24Mesh library in PlatformIO the dependencies RF24 and RF24Network are downloaded and installed correctly. However, RF24Mesh itself is not being installed.

```
[env:nanoatmega328new]
platform = atmelavr
board = nanoatmega328new
framework = arduino

monitor_speed = 115200

lib_deps =
    RF24Mesh@>=1.0.7
```

TMRh20, 2020-05-17:
I'm not sure on this one. I just tested a fresh install of PlatformIO and got errors installing both RF24Mesh and RF24Network. Oddly enough, I was able to install the core RF24 library, then I compiled an RF24Mesh example and somehow it worked. It found the correct libraries and worked??? 

I'm assuming something funky is going on with PlatformIO since we haven't done a new release of RF24Mesh in about a year and this is the first I'm hearing of problems. Might be a good idea to report the issue with PlatformIO via the IDE or https://github.com/platformio

volker, 2021-01-04:
Hello,

I just managed to install RF24, RF24Network, and RF24Mesh (fresh install of PlatformIO) by installing every library explicitly and by referencing their PlatformIO ID, not their name.

`volker@nyx test % pio lib install 433
Library Storage: /Users/volker/projects/iot/test/.pio/libdeps/uno
Library Manager: Installing id:433
Library Manager: RF24 @ 1.3.11 has been installed!
volker@nyx test % pio lib install 435
Library Storage: /Users/volker/projects/iot/test/.pio/libdeps/uno
Library Manager: Installing id:435
Library Manager: RF24Network @ 1.0.14 has been installed!
Library Manager: Installing dependencies...
Library Manager: RF24 @ 1.3.11 is already installed
volker@nyx test % pio lib install 209
Library Storage: /Users/volker/projects/iot/test/.pio/libdeps/uno
Library Manager: Installing id:209
Library Manager: RF24Mesh @ 1.1.4 has been installed!
Library Manager: Installing dependencies...
Library Manager: RF24 @ 1.3.11 is already installed
Library Manager: RF24Network @ 1.0.14 is already installed
volker@nyx test %`

Hope this helps anyone encountering the same errors.

microtronics, 2021-03-12:
I guess this issue resolved itself as I don't encounter it anymore. However RF24@>=1.3.12 appears to not be available in PlatformIO, although the latest changelog is available there...
I'll close this issue

closed 2021-03-12 by microtronics