nRF24/RF24Mesh #174: Master-node communication only works with the first connected node

waleedq, 2020-05-18:
I am working on a network with 12 sensors, still testing with only 3 sensors using the provided master to nodes example, but i am facing the following situation:

- All sensors are sending data to the master successfully and data is received on the master end.
- Sending data from master to other nodes using mesh.write, works only with the first connected node, sending data to other nodes doesn't work from the first time, i need to keep trying to write the data until success.

So basically data flow form node to master is working perfectly but data flow from master to node is only working with the first connected node.
 
Is this a normal case? should i keep trying mesh.write until it's successful?

TMRh20, 2020-05-18:
Hmm, the first thing I would probably suggest is to look at the plain RF24Mesh_Example_Master, and copy the code at the bottom into your sketch that prints out the address list and active nodes. Then verify that your nodes are all connecting properly with unique addresses.  

Then on your nodes, find this piece of code:
  ```   
if ( ! mesh.checkConnection() ) {
   //refresh the network address
  Serial.println("Renewing Address");
  mesh.renewAddress(); 
```

change it to:
```
if( !mesh.getAddress(nodeID) ){
   //refresh the network address
  Serial.println("Renewing Address");
```

Basically, there are some flaws with using mesh.checkConnection, since it only tests communication from node->master. By calling `getAddress` instead, it tests node->master->node. This may be updated or changed in future releases of RF24Mesh.

TMRh20, 2020-05-20:
Been brought to my attention this was incorrect info.

It should be ` if( mesh.getAddress(nodeID) < 0 ){ ` since getAddress returns an int, -1 if not found. Will edit the prev post to correct it... my bad.

waleedq, 2020-05-24:
Hello @TMRh20, thank you for your support, so issue is fixed for me now using both !mesh.checkConnection() and mesh.getAddress(nodeID) < 0, also i've set the checks to be done every ~5 seconds instead of doing it all the time, as i am sending information to the master each time the address is renewed which i guess caused collisions as all the nodes where sending data and kept failing due to the rapid data write.

By limiting the check process to once every 5 seconds the network is very stable now and when the master node is turned off/on again each node renews their address and connect back to the network happily.

Thank you.

closed 2020-05-24 by waleedq