nRF24/RF24Mesh #180: Sensor density per channel

lu4, 2021-02-02:
I've just noticed that `RF24Mesh::begin` method accepts `channel` argument which may mean that the communication  channel is fixed not only for a node but for all network. If so then how many nodes can operate on the same channel in close proximity using the RF24Mesh? According to documentation nRF24L01+ module supports Multiceiver Network concept that supports up to six transmitters and one receiver. Does nRF24L01+ module limitations impact the number of clients that RF24Mesh network can support in close (overlapping) proximity. Does RF24Mesh support arbitrary node to node communication. Does RF24Mesh manage channels somehow to overcome the limitation of clients count on the same channel?

TMRh20, 2021-02-02:
>  If so then how many nodes can operate on the same channel in close proximity using the RF24Mesh?

About 15-25 tops.

> Does RF24Mesh support arbitrary node to node communication.

Yes, it requires nodes to do an address lookup:  https://nrf24.github.io/RF24Mesh/classRF24Mesh.html#a38be0ba49fea6986dedd976f18865912
https://nrf24.github.io/RF24Mesh/classRF24Mesh.html#a895f444dcc0cb175faf845c5bdd7a7d0

> Does RF24Mesh manage channels somehow to overcome the limitation of clients count on the same channel?

No, users can run separate RF24Mesh networks on separate channels if more nodes are required.

lu4, 2021-02-02:
Got, it thank you!

closed 2021-02-02 by lu4