nRF24/RF24Mesh #182: Issue:  DHCP() function Not recognize in Arduino code.

lak1219, 2021-04-01:
Hi,
Anyone can help me?
Calling DHCP() function doesn't work in my Arduino code.  I didn't see the function name orange colour after I checked the code. 

Where is the problem?

Thank you...

[image]

TMRh20, 2021-04-05:
Not sure what isn’t working, it doesn’t matter if it’s orange in the ide. Make sure to test your devices with the basic rf24 examples before using RF24Mesh.

lak1219, 2021-04-05:
Thanks for your response.
Already Basicrf24 and also Rf24mesh working. But the problem is in the DHCP() Function.  I want to call DHCP() function at a specific time because I am going to implement sleep mode to reduce the power consumption of my sensor nodes.

closed 2021-04-07 by lak1219