nRF24/RF24Mesh #190: is EEPROM.h needed in arduino master example?

2bndy5, 2021-07-10:
I don't see where (either in the sketch or in RF24Mesh lib) the EEPROM lib is used (concerning _RF24Mesh_Example_Master.ino_).

I've been skipping this example in the new Arduino CI and PlatformIO CI workflows because of the 
```cpp
#include <EEPROM.h>
```
which is specific to AVR boards.

TMRh20, 2021-07-10:
Nope it’s not used, that is an artifact from testing using eeprom.

2bndy5, 2021-10-30:
@TMRh20 I'm proofreading the docs, and I keep seeing instances where the value for a nodeID ranges [1, 253]. Is this accurate? I would think it ranges [1, 255] (excluding master). This is specific to 

https://github.com/nRF24/RF24Mesh/blob/aa5422bed7302a3148afa7891a097a92d25c363a/RF24Mesh.h#L114-L116
https://github.com/nRF24/RF24Mesh/blob/aa5422bed7302a3148afa7891a097a92d25c363a/RF24Mesh.h#L180-L183

I didn't want to bother with a new thread for a simple question.

TMRh20, 2021-10-31:
I can't remember why, but I thought 254 was reserved for something, and 255 is considered a broadcast IP when using RF24Mesh with an IP stack, so I have it as 253 maximum. It can probably be changed to max 254 or 255, but I'll have to give it some thought and a bit of digging around to see if I can remember wtf.

TMRh20, 2021-10-31:
I can't find anywhere where 254 or 255 is reserved, so I'm pretty confident this can be changed to the range 1-255

2bndy5, 2021-10-31:
Thanks for looking into it. Initially, I thought it might have something to do with the `getNodeID()` return type, but that would only make sense if the returned datatype was int8_t at some historic point. That way -1 & -2 would need to be reserved as 255 & 254, but `getNodeID()` currently returns int16_t (so I had to ask).

commit 656bca17 (2021-07-10): remove artifact; resolves #190
closed 2021-11-14 by TMRh20