nRF24/RF24Mesh #192: Python Wrapper Struct Byte Array Interpreter

hansendm, 2021-07-16:
#184 

So in the previous issue I mentioned that I am trying to pass a C++ struct from my arduino to my raspberry pi. I have a struct that looks like this:

```
struct node_status
{
  char *node_type = "incubator";
  char *sub_type;          // set the type of incubator
  int sub_type_id;
  bool sleep = false;         // set to sleep
  int check_in_time = 1000;   // set check in time
  bool LOCK = false;      // set if admin control is true/false
} nodeStatus;
```
I tried using the python module named struct

```
from RF24 import *
from RF24Network import *
from RF24Mesh import *
import RPi.GPIO as GPIO
import threading
from time import sleep, time
from struct import pack,unpack

radio = RF24(22,0);
network = RF24Network(radio)
mesh = RF24Mesh(radio, network)	
mesh.setNodeID(0) #MasterNode
mesh.begin()
radio.setPALevel(RF24_PA_MAX) # Power Amplifier
radio.printDetails()
		
def millis():
    return int((time()-start)*1000) % (2 ** 32)

def delay(ms):
    ms = ms % (2**32)
    sleep(ms/1000.0)

class RF24_Monitor(threading.Thread):
	def __init__(self):
		print('init')

	def check_network(self):      
		mesh.update()
		mesh.DHCP()
		while network.available():
			print('checking network')
			header, payload = network.read(10)
			nodeID = oct(header.from_node)
			if chr(header.type) == 126:
				print("Rcvd Node Status msg from 0{:o}".format(header.from_node))
				print("node_type: {}".format(unpack("10s",payload[0]))) #node_type
				node_type = unpack("10s",payload[0])
				print("sub_type: {}".format(unpack("10s",payload[1]), header.from_node))	#sub_type
				sub_type = unpack("10s",payload[1])
				print("sub_type_id: {}".format(unpack("b",payload[2])))
				sub_type_id = unpack("b",payload[2])
				print("sleep: {}".format(unpack("?",payload)[3]))	#sleep
				sleep = unpack("?",payload[3])
				print("check_in_time: {}".format(unpack("l",payload[4])))	#check_in_time
				check_in_time = unpack("l",payload[4])
				print("Lock: {}".format(unpack("?",payload[5])))	#LOCK
				Lock = unpack("?",payload[5])
				
			else:
				print("Rcv bad type {} from 0{:o}".format(header.type,header.from_node))
	
	def run(self):
		while True:
			self.check_network()

############################################################
#  Main Entry
############################################################
if __name__ == "__main__":
	RF24_Monitor_Controller = RF24_Monitor()
	RF24_Monitor_Controller.run()
but I am not having much luck. I was even looking at just using ctypes module but seem to not be going anywhere..

from ctypes import *

class interpret_nodes_status(Structure):
    _fields_ = [('node_type',c_char_p),
                ('sub_type',c_char_p),
                ('sub_type_id',c_int),
                ('sleep',c_bool),
                (check_in_time',c_int),
                ('LOCK',c_bool)]

nodestatus = translate_nodes_status(payload)
```
but that just gives me an error

`TypeError: bytes or integer address expected instead of bytearray instance`
What can I do? WHERE am I going wrong with this?

2bndy5, 2021-07-16:
closing as it is a duplicate. @hansendm opening a separate duplicate issue doesn't increase the priority of received notifications on our end. Our responses depend on  how much free time we have and if we can contribute anything towards resolving the issue

I am new blood for the nRF24 org, so I only recently started paying attention to open issues on the RF24Mesh repo.

closed 2021-07-16 by 2bndy5

hansendm, 2021-07-17:
Hey Brendan,

Thanks for doing that. I'm sorry for the duplication. I didn't know if I
was following the right etiquette by keeping my follow up code in the same
post. They were somewhat different topics.

Very Respectfully,
Daniel Hansen

2bndy5, 2021-07-17:
oh. yeah, I'm used to threads getting a little off topic, but your follow up does seem related to the issue's original question.

hansendm, 2021-07-17:
I'm replying to the thread now. I am no pro at this but I really need to
get this thing working. I am at work right now but will be testing out
whatever we figure out here at around 5pm EST (1500L). I hope you can help
me figure this out.

Very Respectfully,
Daniel Hansen