nRF24/RF24Mesh #197: where is duplicate frame detection implemented?

2bndy5, 2021-08-04:
In testing my pure python port, I noticed that sequentially multicasted `NETWORK_POLL` messages using the same `frame.id` were ignored (actually only the first one was handled). I can't find where this condition (to discard duplicate frames) is checked. So, I'm asking to make sure I ported it correctly.

At first I thought it lived in the `network.enqueue()` method, but that only accounts for fragmentation re-assembly. I don't see anything in `network.update()` or `mesh.update()` that discards frames with a duplicate ID number.

TMRh20, 2021-08-04:
I don’t remember there being any functionality to drop duplicate frames except maybe in fragmentation/reassembly. I could be wrong but I’d have to scour the code to find out for sure. 
Are you sure about the consistency of this behaviour?

2bndy5, 2021-08-04:
Yes, I was running a C++ master node on RPi and trying to get CirPy non-master node on RP2040 MCU to connect to it. 

At the time, my lib was only incrementing the frame ID number for user payloads. The master node on RPi would only respond to the first Poll, then ignore all subsequent frame's with the same ID. As soon as I started incrementing the frame's ID (from the CirPy node), the C++ master node started responding to each Poll message.

This would take a bit of work to reproduce...

2bndy5, 2021-08-04:
This behavior is also noted in the [muticastRelay docs](https://nrf24.github.io/RF24Network/classRF24Network.html#add7408ca2b2f9ab15331ba77a747f7bb)

TMRh20, 2021-08-04:
Hmm I’ll take your word for it, and take a closer look.

2bndy5, 2021-08-05:
Nevermind, I tried to reproduce this, and the C++ master node responded to each Poll message when the frame's id number remained consistent. When I first noticed the above scenario, I was trying to use a python version of `writeFast()` and `txStandby()`... I've since reverted back to only using 1 level of the TX FIFO for reliability reasons. So, the duplicate frame's may actually have never been properly transmitted at the time.

I'm still curious how `multicastRelay` filters out duplicate frames though. Is it only done on the receiving nodes?

TMRh20, 2021-08-05:
I still can’t seem to find where it would filter out duplicate frames anywhere. The docs may be incorrect.

2bndy5, 2021-08-05:
Could it be referring to inherent collision during multicasting?

TMRh20, 2021-08-05:
I’m thinking I intended to include duplicate filtering but decided against it…

2bndy5, 2021-08-05:
It may be too much trying for too little benefit... You'd have to keep a cache of the last handled header separate from the `frame_buffer`; and that only solves sequential duplicate frames.

I take that part out of the docs.

closed 2021-08-05 by 2bndy5