nRF24/RF24Mesh #198: Mesh examples fail, RF24Mesh_Example.ino fails to mesh.begin()

otmezger, 2021-08-08:
Hi, 

I'm having some trouble runnign the examples. 

Setup: 
Nano 1 running RF24Mesh_Example_Master.ino. 
Nano 2 running RF24Mesh_Example.ino 

On the Master, I get the Assigned Addresses correctly like this:

```
********Assigned Addresses********
NodeID: 1 RF24Network Address: 05
**********************************

```

However, on the node, `mesh.begin(channel)` returns 0 on the setup, and after a few random amount of `Renewing Address` it starts sending the millis. It then breaks again after a random amount of time and is stuck in `Renewing Address`. 

I'm setting `#define channel 97 // The radio channel (1-127) default:97` on both Master and Node. 

Please note that I tested with the example StreamingData.ino from the RF24 library, and it worked correctly with the same hardware. 

Could this be a power problem? I noticed that the StreamingData.ino has the following lines:

```
radio.setPALevel(RF24_PA_LOW); // RF24_PA_MAX is default
```

However, if I set this in the Node in setup, the program is stuck there and never continues. I never get the `Done` message printed. 

```
void setup() {

  Serial.begin(115200);
  Serial.print("attemptng to set PALevel: ");
  radio.setPALevel(RF24_PA_LOW); // RF24_PA_MAX is default
  Serial.print(" Done");
  //printf_begin();
  // Set the nodeID manually
  mesh.setNodeID(nodeID);
  // Connect to the mesh
  Serial.println(F("Connecting to the mesh..."));
  //   bool meshBegin = mesh.begin(channel, data_rate, timeout);
  bool meshBegin = mesh.begin(channel);
  Serial.println(meshBegin);  
  
}
```

What would be the proper way to debug this? How can I set the safe power mode using Mesh? 

thanks

Avamander, 2021-08-08:
Start with diagnosing the lower layers first. Try the RF24's examples, such as pingpair_dyn or transfer.

otmezger, 2021-08-08:
Thanks @Avamander for your quick answer. As mentioned, I tried the example StreamingData.ino from the RF24 library and it worked perfectly.  

will try the other you mentioned and see if I find something.

otmezger, 2021-08-08:
Dear @Avamander  I just tried the pingpair_dyn example and it worked perfectly out of the box. Here is the output I got on the pig setting:

```
RF24/examples/pingpair_dyn/
ROLE: Ping out
SPI Speedz	= 10 Mhz
STATUS		= 0x0e RX_DR=0 TX_DS=0 MAX_RT=0 RX_P_NO=7 TX_FULL=0
RX_ADDR_P0-1	= 0xf0f0f0f0e1 0xf0f0f0f0d2
RX_ADDR_P2-5	= 0xc3 0xc4 0xc5 0xc6
TX_ADDR		= 0xf0f0f0f0e1
RX_PW_P0-6	= 0x20 0x20 0x20 0x20 0x20 0x20
EN_AA		= 0x3f
EN_RXADDR	= 0x02
RF_CH		= 0x4c
RF_SETUP	= 0x07
CONFIG		= 0x0f
DYNPD/FEATURE	= 0x3f 0x04
Data Rate	= 1 MBPS
Model		= nRF24L01+
CRC Length	= 16 bits
PA Power	= PA_MAX
ARC		= 0
Now sending length 4
Got response size=4 value=ABCD
Now sending length 5
Got response size=5 value=ABCDE
Now sending length 6
Got response size=6 value=ABCDEF
Now sending length 7
Got response size=7 value=ABCDEFG
Now sending length 8
Got response size=8 value=ABCDEFGH
Now sending length 9
Got response size=9 value=ABCDEFGHI
Now sending length 10
Got response size=10 value=ABCDEFGHIJ
Now sending length 11
Got response size=11 value=ABCDEFGHIJK
Now sending length 12
Got response size=12 value=ABCDEFGHIJKL
Now sending length 13
Got response size=13 value=ABCDEFGHIJKLM
Now sending length 14
Got response size=14 value=ABCDEFGHIJKLMN
Now sending length 15
Got response size=15 value=ABCDEFGHIJKLMNO
Now sending length 16
Got response size=16 value=ABCDEFGHIJKLMNOP
Now sending length 17
Got response size=17 value=ABCDEFGHIJKLMNOPQ
Now sending length 18
Got response size=18 value=ABCDEFGHIJKLMNOPQR
Now sending length 19
Got response size=19 value=ABCDEFGHIJKLMNOPQRS
Now sending length 20
Got response size=20 value=ABCDEFGHIJKLMNOPQRST
Now sending length 21
Got response size=21 value=ABCDEFGHIJKLMNOPQRSTU
Now sending length 22
Got response size=22 value=ABCDEFGHIJKLMNOPQRSTUV
Now sending length 23
Got response size=23 value=ABCDEFGHIJKLMNOPQRSTUVW
Now sending length 24
Got response size=24 value=ABCDEFGHIJKLMNOPQRSTUVWX
Now sending length 25
Got response size=25 value=ABCDEFGHIJKLMNOPQRSTUVWXY
Now sending length 26
Got response size=26 value=ABCDEFGHIJKLMNOPQRSTUVWXYZ
Now sending length 27
Got response size=27 value=ABCDEFGHIJKLMNOPQRSTUVWXYZ7
Now sending length 28
Got response size=28 value=ABCDEFGHIJKLMNOPQRSTUVWXYZ78
Now sending length 29
Got response size=29 value=ABCDEFGHIJKLMNOPQRSTUVWXYZ789
Now sending length 30
Got response size=30 value=ABCDEFGHIJKLMNOPQRSTUVWXYZ7890
Now sending length 31
Got response size=31 value=ABCDEFGHIJKLMNOPQRSTUVWXYZ78901
Now sending length 32
Got response size=32 value=ABCDEFGHIJKLMNOPQRSTUVWXYZ789012
Now sending length 4
Got response size=4 value=ABCD
Now sending length 5
Got response size=5 value=ABCDE
Now sending length 6
Got response size=6 value=ABCDEF
Now sending length 7
Got response size=7 value=ABCDEFG
Now sending length 8
Got response size=8 value=ABCDEFGH
Now sending length 9
Got response size=9 value=ABCDEFGHI
Now sending length 10
Got response size=10 value=ABCDEFGHIJ
Now sending length 11
Got response size=11 value=ABCDEFGHIJK
Now sending length 12
Got response size=12 value=ABCDEFGHIJKL
Now sending length 13
Got response size=13 value=ABCDEFGHIJKLM
Now sending length 14
Got response size=14 value=ABCDEFGHIJKLMN
Now sending length 15
Got response size=15 value=ABCDEFGHIJKLMNO
Now sending length 16
Got response size=16 value=ABCDEFGHIJKLMNOP
Now sending length 17
Got response size=17 value=ABCDEFGHIJKLMNOPQ
Now sending length 18
Got response size=18 value=ABCDEFGHIJKLMNOPQR
Now sending length 19
Got response size=19 value=ABCDEFGHIJKLMNOPQRS
Now sending length 20
Got response size=20 value=ABCDEFGHIJKLMNOPQRST
Now sending length 21
Got response size=21 value=ABCDEFGHIJKLMNOPQRSTU
Now sending length 22
Got response size=22 value=ABCDEFGHIJKLMNOPQRSTUV
Now sending length 23
Got response size=23 value=ABCDEFGHIJKLMNOPQRSTUVW
Now sending length 24
Got response size=24 value=ABCDEFGHIJKLMNOPQRSTUVWX
Now sending length 25
Got response size=25 value=ABCDEFGHIJKLMNOPQRSTUVWXY
Now sending length 26
Got response size=26 value=ABCDEFGHIJKLMNOPQRSTUVWXYZ
Now sending length 27
Got response size=27 value=ABCDEFGHIJKLMNOPQRSTUVWXYZ7
Now sending length 28
Got response size=28 value=ABCDEFGHIJKLMNOPQRSTUVWXYZ78
Now sending length 29
Got response size=29 value=ABCDEFGHIJKLMNOPQRSTUVWXYZ789
Now sending length 30
Got response size=30 value=ABCDEFGHIJKLMNOPQRSTUVWXYZ7890
Now sending length 31
Got response size=31 value=ABCDEFGHIJKLMNOPQRSTUVWXYZ78901
Now sending length 32
Got response size=32 value=ABCDEFGHIJKLMNOPQRSTUVWXYZ789012
```

However, I'm still unable to reliably get the mesh to work, `mesh.begin(channel)` will mostly return `false` on the node. Any other ideas?

TMRh20, 2021-08-08:
Try starting the mesh on a different channel maybe.

otmezger, 2021-08-09:
Thanks @TMRh20, will try. I already tried with 2 channels but will try with more. maybe doing a network scan and ensure to take an empty channel helps.

otmezger, 2021-08-30:
I finally had some time last weekend and found that not all (empty) channels will work as expected. It worked in the lower end of the channels. Thanks.

closed 2021-11-06 by TMRh20