nRF24/RF24Mesh #199: undefined symbol error in librf24mesh.so

pprakash, 2021-11-28:
I am trying to test out RF24 modules with Rapberry Pi Zero W and Arduino Uno, using Python on the Pi side. Have been able to successfully compile and test RF24 and RF24Network functionalities. While I started working on RF24Mesh example, I am getting below error while importing the library. I tried searching for a solution but couldn't find one.  Any help in this regard?

I am using 1.1.6 version of the library.

Error:

`Python 3.9.2 (default, Mar 12 2021, 04:06:34)
[GCC 10.2.1 20210110] on linux
Type "help", "copyright", "credits" or "license" for more information.
>>> from RF24Mesh import *
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
**ImportError: /usr/local/lib/librf24mesh.so.1: undefined symbol: _ZN17RF24NetworkHeader7next_idE**
>>> from RF24Network import *
>>> from RF24 import *
>>>
`

2bndy5, 2021-11-28:
Did you reinstall the other RF24 libraries?
How did you install v1.1.6? (I assume you built it from source)
Can you post your results from executing
```
nm -gDC /usr/local/lib/librf24mesh.so
```
it might be best if delete the old librf24*.so from _/usr/local/lib/_. And reinstall the libs (and python wrappers) in order of OSI layer (RF24->RF24Network->RF24Mesh).

2bndy5, 2021-11-29:
I can't reproduce this. I need more info about context.

pprakash, 2021-11-29:
> 1. Did you reinstall the other RF24 libraries? We just did a complete release crusade for most RF24* libs.
> 2. How did you install v1.1.6? (I assume you built it from source)
> 3. Can you post your results from executing
>    ```
>    nm -gDC /usr/local/lib/librf24mesh.so
>    ```
> 
> It might be best if you delete the old librf24*.so from _/usr/local/lib/_. And reinstall the libs (and python wrappers) in order of OSI layer (RF24->RF24Network->RF24Mesh).
> 
> I had to add the linking of RF24Network and RF24 in the python wrapper build process for RF24Mesh because I couldn't get v1.1.5 to work.

I have followed Automated Install instructions from [https://nrf24.github.io/RF24/md_docs_linux_install.html](url). After the above error, I downloaded the v1.1.5 from [https://www.arduino.cc/reference/en/libraries/rf24mesh/](url) and did a manual install as per below steps of only RF24Mesh and it started working for me with a pair of Uno slaves/nodes.

Manual Steps followed (given only the commands which I issued here).
`pi@raspberrypi:~/RF24 $ wget https://downloads.arduino.cc/libraries/github.com/TMRh20/RF24Mesh-1.1.5.zip
pi@raspberrypi:~/RF24 $ unzip RF24Mesh-1.1.5.zip
pi@raspberrypi:~/RF24 $ cd RF24Mesh-1.1.5/pyRF24Mesh/
pi@raspberrypi:~/RF24/RF24Mesh-1.1.5/pyRF24Mesh $ python setup.py build
pi@raspberrypi:~/RF24/RF24Mesh-1.1.5/pyRF24Mesh $ sudo python setup.py install
`

nm -gDC /usr/local/lib/librf24mesh.so output - This is from v1.1.5.

`pi@raspberrypi:~ $ nm -gDC /usr/local/lib/librf24mesh.so
         U __aeabi_unwind_cpp_pr1@GCC_3.5
         U __cxa_end_cleanup@CXXABI_1.3
         w __cxa_finalize@GLIBC_2.4
         w __gmon_start__
         U __gxx_personality_v0@CXXABI_1.3
         w _ITM_deregisterTMCloneTable
         w _ITM_registerTMCloneTable
         U malloc@GLIBC_2.4
         U memcpy@GLIBC_2.4
         U __millis
         U __msleep
         U realloc@GLIBC_2.4
         U RF24Network::begin(unsigned char, unsigned short)
00002e34 W RF24Network::begin(unsigned short)
         U RF24Network::write(RF24NetworkHeader&, void const*, unsigned short)
         U RF24Network::write(RF24NetworkHeader&, void const*, unsigned short, unsigned short)
         U RF24Network::update()
         U RF24Network::multicast(RF24NetworkHeader&, void const*, unsigned short, unsigned char)
         U RF24NetworkHeader::next_id
00002db0 W RF24NetworkHeader::RF24NetworkHeader(unsigned short, unsigned char)
00002d8c W RF24NetworkHeader::RF24NetworkHeader()
00002db0 W RF24NetworkHeader::RF24NetworkHeader(unsigned short, unsigned char)
00002d8c W RF24NetworkHeader::RF24NetworkHeader()
         U RF24::setChannel(unsigned char)
         U RF24::setDataRate(rf24_datarate_e)
         U RF24::stopListening()
         U RF24::startListening()
         U RF24::begin()
         U RF24::available()
00001b24 T RF24Mesh::getAddress(unsigned char)
000026cc T RF24Mesh::setAddress(unsigned char, unsigned short, bool)
000019d4 T RF24Mesh::setChannel(unsigned char)
00002d5c T RF24Mesh::setCallback(void (*)())
00001f54 T RF24Mesh::beginDefault()
0000202c T RF24Mesh::renewAddress(unsigned int)
00001fa4 T RF24Mesh::releaseAddress()
000021d0 T RF24Mesh::requestAddress(unsigned char)
00001a9c T RF24Mesh::checkConnection()
0000268c T RF24Mesh::setStaticAddress(unsigned char, unsigned short)
00002ab8 T RF24Mesh::DHCP()
00001484 T RF24Mesh::begin(unsigned char, rf24_datarate_e, unsigned int)
000018b0 T RF24Mesh::write(void const*, unsigned char, unsigned int, unsigned char)
00001824 T RF24Mesh::write(unsigned short, void const*, unsigned char, unsigned int)
0000161c T RF24Mesh::update()
00001ef8 T RF24Mesh::getLevel(unsigned short)
000028a8 T RF24Mesh::loadDHCP()
000029f8 T RF24Mesh::saveDHCP()
00001a30 T RF24Mesh::setChild(bool)
00001d0c T RF24Mesh::getNodeID(unsigned short)
00002658 T RF24Mesh::setNodeID(unsigned char)
00001438 T RF24Mesh::RF24Mesh(RF24&, RF24Network&)
00001438 T RF24Mesh::RF24Mesh(RF24&, RF24Network&)
00002e9c W std::fpos<__mbstate_t>::operator long long() const
         U std::basic_ios<char, std::char_traits<char> >::operator!() const@GLIBCXX_3.4
         U std::istream::read(char*, int)@GLIBCXX_3.4
         U std::istream::seekg(long long, std::_Ios_Seekdir)@GLIBCXX_3.4
         U std::istream::tellg()@GLIBCXX_3.4
         U std::ostream::write(char const*, int)@GLIBCXX_3.4
         U std::basic_ifstream<char, std::char_traits<char> >::close()@GLIBCXX_3.4
         U std::basic_ifstream<char, std::char_traits<char> >::basic_ifstream(char const*, std::_Ios_Openmode)@                                                                                                                   GLIBCXX_3.4
         U std::basic_ifstream<char, std::char_traits<char> >::~basic_ifstream()@GLIBCXX_3.4
         U std::basic_ofstream<char, std::char_traits<char> >::close()@GLIBCXX_3.4
         U std::basic_ofstream<char, std::char_traits<char> >::basic_ofstream(char const*, std::_Ios_Openmode)@                                                                                                                   GLIBCXX_3.4
         U std::basic_ofstream<char, std::char_traits<char> >::~basic_ofstream()@GLIBCXX_3.4
00002e6c W std::operator|(std::_Ios_Openmode, std::_Ios_Openmode)`

pprakash, 2021-11-29:
**Thanks @2bndy5 **
UPDATE: Have removed the librf24* files from /usr/local/lib and re-installed the library as per automated install instruction from [https://nrf24.github.io/RF24/md_docs_linux_install.html](url).

Only change which I did this time is that I have build and installed RF24, RF24Network and RF24Mesh modules individually for Python. 

It works without any error now. Attached the installation log for reference.
[RF24,RF24Network,RF24Mesh Install.txt](https://github.com/nRF24/RF24Mesh/files/7619133/RF24.RF24Network.RF24Mesh.Install.txt)

2bndy5, 2021-11-29:
The only thing I would object to is that you seem to use python 2 to install the python wrappers. Python 2 hit end-of-life in January 2020. While we technically still support it, many other tools stopped supporting python 2. We will also stop supporting it when boost.python stops supporting it.

From now on I recommend you use `python3` or use a virtual environment (venv for short)
```
python3 -m venv env
source env/bin/activate
```
This way issuing `python` commands (while the venv is activated) will use python 3. Furthermore, you wouldn't need to use `sudo` to install (`sudo python` will have unintended consequences inside an activated venv).

pprakash, 2021-11-29:
Thank you, the default installation didnt had Python 2.x and hence I didn't explicitly called python3.
`
pi@raspberrypi:~ $ python --version
Python 3.9.2
pi@raspberrypi:~ $
`

Noted on the "sudo python" part.

closed 2021-11-29 by 2bndy5

Lewatoto, 2022-01-29:
Hello I´m having the same issue with the instalation of python wrappers, already remove all content of /usr/local/lib, sunr the install.sh and followed the order RF24, RF24Network and RF24Mesh for the python installation but have the same error when tried to run some RF24Mesh python example, when I compile the cpp all works ok.

Raspberry pi 2 B - Raspbian GNU/Linux 11 (bullseye)

RF24              = 1.4.2
RF24Network = 1.0.16
RF24Mesh      = 1.1.6

The error showed on the console:
```
pi@estacion:~/rf24libs/RF24Mesh/examples_RPi $ sudo python RF24Mesh_Example_Master.py 
Traceback (most recent call last):
  File "/home/pi/rf24libs/RF24Mesh/examples_RPi/RF24Mesh_Example_Master.py", line 7, in <module>
    from RF24Mesh import *
ImportError: /usr/local/lib/librf24mesh.so.1: undefined symbol: _ZN17RF24NetworkHeader7next_idE
```

The output of `nm -gDC /usr/local/lib/librf24mesh.so`:
```
pi@estacion:~/rf24libs/RF24Mesh/examples_RPi $ nm -gDC /usr/local/lib/librf24mesh.so
         U __aeabi_unwind_cpp_pr1@GCC_3.5
         U __cxa_end_cleanup@CXXABI_1.3
         w __cxa_finalize@GLIBC_2.4
         w __gmon_start__
         U __gxx_personality_v0@CXXABI_1.3
         w _ITM_deregisterTMCloneTable
         w _ITM_registerTMCloneTable
         U malloc@GLIBC_2.4
         U memcpy@GLIBC_2.4
         U __millis
         U __msleep
         U realloc@GLIBC_2.4
         U RF24Network::begin(unsigned char, unsigned short)
00002e34 W RF24Network::begin(unsigned short)
         U RF24Network::write(RF24NetworkHeader&, void const*, unsigned short)
         U RF24Network::write(RF24NetworkHeader&, void const*, unsigned short, unsigned short)
         U RF24Network::update()
         U RF24Network::multicast(RF24NetworkHeader&, void const*, unsigned short, unsigned char)
         U RF24NetworkHeader::next_id
00002db0 W RF24NetworkHeader::RF24NetworkHeader(unsigned short, unsigned char)
00002d8c W RF24NetworkHeader::RF24NetworkHeader()
00002db0 W RF24NetworkHeader::RF24NetworkHeader(unsigned short, unsigned char)
00002d8c W RF24NetworkHeader::RF24NetworkHeader()
         U RF24::setChannel(unsigned char)
         U RF24::setDataRate(rf24_datarate_e)
         U RF24::stopListening()
         U RF24::startListening()
         U RF24::begin()
         U RF24::available()
00001b24 T RF24Mesh::getAddress(unsigned char)
000026cc T RF24Mesh::setAddress(unsigned char, unsigned short, bool)
000019d4 T RF24Mesh::setChannel(unsigned char)
00002d5c T RF24Mesh::setCallback(void (*)())
00001f54 T RF24Mesh::beginDefault()
0000202c T RF24Mesh::renewAddress(unsigned int)
00001fa4 T RF24Mesh::releaseAddress()
000021d0 T RF24Mesh::requestAddress(unsigned char)
00001a9c T RF24Mesh::checkConnection()
0000268c T RF24Mesh::setStaticAddress(unsigned char, unsigned short)
00002ab8 T RF24Mesh::DHCP()
00001484 T RF24Mesh::begin(unsigned char, rf24_datarate_e, unsigned int)
000018b0 T RF24Mesh::write(void const*, unsigned char, unsigned int, unsigned char)
00001824 T RF24Mesh::write(unsigned short, void const*, unsigned char, unsigned int)
0000161c T RF24Mesh::update()
00001ef8 T RF24Mesh::getLevel(unsigned short)
000028a8 T RF24Mesh::loadDHCP()
000029f8 T RF24Mesh::saveDHCP()
00001a30 T RF24Mesh::setChild(bool)
00001d0c T RF24Mesh::getNodeID(unsigned short)
00002658 T RF24Mesh::setNodeID(unsigned char)
00001438 T RF24Mesh::RF24Mesh(RF24&, RF24Network&)
00001438 T RF24Mesh::RF24Mesh(RF24&, RF24Network&)
00002e9c W std::fpos<__mbstate_t>::operator long long() const
         U std::basic_ios<char, std::char_traits<char> >::operator!() const@GLIBCXX_3.4
         U std::istream::read(char*, int)@GLIBCXX_3.4
         U std::istream::seekg(long long, std::_Ios_Seekdir)@GLIBCXX_3.4
         U std::istream::tellg()@GLIBCXX_3.4
         U std::ostream::write(char const*, int)@GLIBCXX_3.4
         U std::basic_ifstream<char, std::char_traits<char> >::close()@GLIBCXX_3.4
         U std::basic_ifstream<char, std::char_traits<char> >::basic_ifstream(char const*, std::_Ios_Openmode)@GLIBCXX_3.4
         U std::basic_ifstream<char, std::char_traits<char> >::~basic_ifstream()@GLIBCXX_3.4
         U std::basic_ofstream<char, std::char_traits<char> >::close()@GLIBCXX_3.4
         U std::basic_ofstream<char, std::char_traits<char> >::basic_ofstream(char const*, std::_Ios_Openmode)@GLIBCXX_3.4
         U std::basic_ofstream<char, std::char_traits<char> >::~basic_ofstream()@GLIBCXX_3.4
00002e6c W std::operator|(std::_Ios_Openmode, std::_Ios_Openmode)
```

I attach a file with all the commands and the order that I´ve follow.

[RF24_installation.txt](https://github.com/nRF24/RF24Mesh/files/7963129/RF24_installation.txt)

2bndy5, 2022-01-29:
```
In file included from /usr/include/boost/smart_ptr/detail/sp_thread_sleep.hpp:22,
                 from /usr/include/boost/smart_ptr/detail/yield_k.hpp:23,
                 from /usr/include/boost/smart_ptr/detail/spinlock_gcc_atomic.hpp:14,
                 from /usr/include/boost/smart_ptr/detail/spinlock.hpp:42,
                 from /usr/include/boost/smart_ptr/detail/spinlock_pool.hpp:25,
                 from /usr/include/boost/smart_ptr/shared_ptr.hpp:29,
                 from /usr/include/boost/shared_ptr.hpp:17,
                 from /usr/include/boost/python/converter/shared_ptr_to_python.hpp:12,
                 from /usr/include/boost/python/converter/arg_to_python.hpp:15,
                 from /usr/include/boost/python/call.hpp:15,
                 from /usr/include/boost/python/object_core.hpp:14,
                 from /usr/include/boost/python/args.hpp:22,
                 from /usr/include/boost/python.hpp:11,
                 from pyRF24.cpp:1:
/usr/include/boost/bind.hpp:36:1: note: ‘#pragma message: The practice of declaring the Bind placeholders (_1, _2, ...) in the global namespace is deprecated. Please use <boost/bind/bind.hpp> + using namespace boost::placeholders, or define BOOST_BIND_GLOBAL_PLACEHOLDERS to retain the current behavior.’
   36 | BOOST_PRAGMA_MESSAGE(
      | ^~~~~~~~~~~~~~~~~~~~
/usr/include/boost/detail/iterator.hpp:13:1: note: ‘#pragma message: This header is deprecated. Use <iterator> instead.’
   13 | BOOST_HEADER_DEPRECATED("<iterator>")
      | ^~~~~~~~~~~~~~~~~~~~~~~
```

I noticed this deprecation message as well, but it hasn't caused any problems every time I install the wrappers.

Another possible problem is if the python wrappers already installed, then the belly built python wrappers may not install. Meaning the old python wrappers would try to be linking to newly installed C++ libs.

Try uninstalling everything including the portion wrappers using what you've done already plus the command
```
pip3 uninstall RF24 RF24Network RF24Mesh
```
It looks like the default version invoked by `python` (& hopefully `pip`) is now calling python3 on bullseye.

The other thing I tend to do is use a python venvq, so I can more easily uninstall the python wrappers (I've run into errors trying to do so in the past). Additionally, using a python venv does not require sudo to install (which is good).

2bndy5, 2022-01-29:
ok I was finally able to reproduce this problem (for RF24Network and RF24Mesh)... I fixed it on the master branches by reversing the specified linking order of rf24 libs.

1. delete everything in rf24libs folder: `sudo rm -r ~/rf24libs/*`
2. delete old .so files: `sudo rm /usr/local/lib/librf24*`
3. delete previously install python wrappers: `pip uninstall RF24 RF24Network RF24Mesh`
4. run the installer script again and re-install python wrappers

step 5 should pull in the fixes I just pushed to the master branches

Lewatoto, 2022-01-29:
many thanks, now in my tests all work ok.