nRF24/RF24Mesh #203: add option to reattempt connection in setup() for arduino examples

2bndy5, 2022-02-18:
In testing the mesh examples on arduino, I found it kinda annoying to get up and reset the MCU when the examples don't connect during `setup()`. This is completely my fault since I added an infinite loop when `mesh.begin()` returned false (for unresponsive radio hardware reasons).

### Proposal
For any non-master node examples, change
```cpp
  // Connect to the mesh
  Serial.println(F("Connecting to the mesh..."));
  if (!mesh.begin()) {
    Serial.println(F("Radio hardware not responding or could not connect to network."));
    while (1) {
      // hold in an infinite loop
    }
  }
```
to
```cpp
  // Connect to the mesh
  Serial.println(F("Connecting to the mesh..."));
  if (!mesh.begin()) {
    if (radio.isChipConnected()) {
      while (mesh.mesh_address == MESH_DEFAULT_ADDRESS) {
        Serial.println(F("Could not connect to network."));
        Serial.println(F("Press any key to reattempt a connection to the mesh."));
        while (!Serial.available()) {
          // wait for user input
        }
        Serial.read(); // dispose of first input char
        Serial.println(F("Connecting to the mesh..."));
        mesh.renewAddress();
      }
    }
    else {
      Serial.println(F("Radio hardware not responding."));
      while (1) {
        // hold in an infinite loop
      }
    }
  }
```

TMRh20, 2022-02-18:
I think this is a good idea, however I'm not sure we need to wait for serial input to try again, why not just loop the renew in automatically?

2bndy5, 2022-02-18:
good idea. I'll adjust it to do that.

closed 2022-03-04 by TMRh20