nRF24/RF24Mesh #206: Is there a use case for a changing master node?

2bndy5, 2022-02-19:
I was testing the exposed `addrList` in pyrf24 and hit a segmentation fault when I changed the mesh child node into a master node and explicitly called `loadDHCP()`. I already can see the problem because if the node's ID is not 0 upon calling `mesh.begin()`, then the `addrList` doesn't have any memory allocated.

Is there a practical use case in which the master node of a mesh can change similar to how child nodes change?

This wouldn't be a hard problem to solve, but it would definitely involve increased compile size...

<details><summary>code to reproduce</summary>

```cpp
mesh.setNodeID(1);
mesh.begin();
mesh.setNodeID(0);
mesh.renewAddress();
mesh.loadDHCP();
```
</details>

TMRh20, 2022-02-21:
I don't think there is really a use case for it, plus as you mentioned it would increase memory and program space usage just to support an unlikely scenario.

2bndy5, 2022-02-21:
I also had an idea a while back (when porting to pure python) to reserve a msg type that transfers the DHCP list to a new master node. The only case for that would be if you want a continuous network running while phasing in a firmware update.

closed 2022-02-21 by 2bndy5