nRF24/RF24Mesh #221: move `addrList` allocation to c'tor

2bndy5, 2023-02-25:
I'm only pursuing this proposal because it would obsolete the private member `addrMemAllocated` member. Initially, this idea came to me (in #220) when discovering that static addresses have to be assigned _after_ calling `begin()`. In order for a network administrator to give priority to network addresses, they'd have to assign the addresses _before_ calling `update()` (this proposal won't change that). However, the requirement for calling `begin()` feels a bit cumbersome as the `addrList` array has nothing to do with initializing the required hardware.

I can't think of a reason to keep the allocating code in `begin()`, so maybe there's a historical reason about why it was written the way it is now.

The proposed code change is pretty straight forward as mentioned in the issue title.

2bndy5, 2023-02-25:
> I can't think of a reason to keep the allocating code in begin()

I just thought of a reason: Allocating the array is dependent on the ID number passed to `setNodeID()`. I'm not sure if this proposal is feasible now.

2bndy5, 2023-02-25:
Maybe we can move the allocating code into `setNodeID()`. As for obsoleting the `addrMemAllocated` member, we might be able to check if the pointer is null instead.

TMRh20, 2023-02-26:
I think those are probably both good ideas.