nRF24/RF24Mesh #225: Using Mesh and Broadcast?

mchacher, 2023-07-31:
Hi, I am about to use this great library in a mesh context.
I would have 2 questions:
- how are the "nrf24" tree levels used when using mesh library, since nodes are only defined with a unique number ranging from 1 to 255?
- can we leverage broadcast communication from the master to the nodes? How to send such a broadcast message? Can you provide a simple example?

2bndy5, 2023-07-31:
1. The master node assigns the connecting node a network (logical) address. The network address informs the other nodes how the newly connected mesh node fits within the network topology. See our documented [Mesh connecting process](https://nrf24.github.io/pyRF24/topology.html#rf24mesh-connecting-process) for more detail.
2. Using [`network.multicast()`](https://nrf24.github.io/RF24Network/classESBNetwork.html#a3e6e9e74461290d6f648e972ee4aaf91) seems to be what you're asking about, unless I misunderstood the question. That doc link also references other multicast-related functionality (in the "see also" section).
   > Can you provide a simple example?

   Please read the docs as this functionality is pretty straight forward once you understand the network topology.

Both of your questions seem related to the network topology. The [pyrf24/topology](https://nrf24.github.io/pyRF24/topology.html) page has some good basic information, but it uses the python representation of octal numbers (`0o0`) instead of the C++ representation of octal numbers (`00`). 

> unique number ranging from 1 to 255

Please keep in mind that a network can only have 256 mesh nodes (including master node). Using a mix of statically assigned nodes (RF24Network) with mesh nodes is allowed. See [`mesh.setAddress()`](https://nrf24.github.io/RF24Mesh/classESBMesh.html#aef94c82c204d9c2c14c493687e37fc40) about giving static RF24Network nodes a unique RF24Mesh ID.

TMRh20, 2023-07-31:
Just to add to what Brendan posted, multicast doesn’t really work with RF24Mesh since you don’t want to enable multicast forwarding or enable all nodes to use the same multicast level. This is because the mesh uses multicast to find other devices when connecting.

closed 2024-04-27 by TMRh20