nRF24/RF24Mesh #227: issues using core 0 on esp32.

ripjohnbrown1859, 2023-09-20:
i am using an esp32 and need to run the networking on core 0 as I have time critical stuff running on core 1. 
node
```
void networkTaskCode( void * pvParameters ){
  radio.setPALevel(RF24_PA_MIN);
  if (!mesh.begin()) {
    if (radio.isChipConnected()) {
      do {
        // mesh.renewAddress() will return MESH_DEFAULT_ADDRESS on failure to connect
        latestMessage = ("connecting to mesh");
      } while (mesh.renewAddress() == MESH_DEFAULT_ADDRESS);
    } else {
      Serial.println(F("Radio hardware not responding."));

    }
  }

  for(;;){
   
    
   mesh.update();
   
    if (!mesh.checkConnection()) {
  //refresh the network address
  latestMessage = ("Renewing Address");
  if (mesh.renewAddress() == MESH_DEFAULT_ADDRESS) {
      //If address renewal fails, reconfigure the radio and restart the mesh
      //This allows recovery from most if not all radio errors
      mesh.begin();
      latestMessage = ("Failed to renew Address");}
    }else{
      latestMessage = ("Connected");
    }

    RF24NetworkHeader header;
    payload = {systolic, diastolic, lastValidSpO2, lastValidBPM, irBuffer[1]};
    
    if(!mesh.write(&payload,77, sizeof(payload))){
      latestMessage = ("failed to send");
    }else{
      latestMessage = ("sent");
    }
  yield();
  }
}
```
master
```
  mesh.update();
  mesh.DHCP();
  btn.tick();
  newPos = encoder.getCount()/2; 
  if(network.available()){
    latestMessage = "message received";
    RF24NetworkHeader header;
    //payload_t payload;
    network.read(header, &payload, sizeof(payload));

  }
```
im getting a mesh connection but failing to send any data between the node and the master.

2bndy5, 2023-09-21:
The code you show seems fine. Maybe there's too much going on in the code you didn't show. I suspect there might be a timing issue between when the node `mesh::write()` and the master `mesh.update()`. But we really don't have much to go on here.

ripjohnbrown1859, 2023-09-21:
heres the whole loop for the master.
```
oid loop() {
  // put your main code here, to run repeatedly:
  mesh.update();
  mesh.DHCP();
  btn.tick();
  u8g2.clearBuffer();
  newPos = encoder.getCount()/2; 
  if(network.available()){
    latestMessage = "message received";
    RF24NetworkHeader header;
    network.peek(header);
    payload_t payload;
    switch(header.type){
      case 77:
      network.read(header, &payload, sizeof(payload));
      u8g2.drawStr(0,50,String(payload.lastValidSpO2).c_str());
      break;
      default:
      network.read(header, 0, 0);
      break;
    }
  }
  if(newPos > oldPos){
    selector++;
    if(selector > selectorMax){
      selector = selectorMin;
    }
  }
  if(newPos < oldPos){
    selector--;
    if(selector < selectorMin||selector == 255){
      selector = selectorMax;
    }
  }
  u8g2.drawStr(0,10,WiFi.macAddress().c_str());
  u8g2.drawStr(0,20,String(selector).c_str());
  u8g2.drawStr(0,30,latestMessage.c_str());
  u8g2.drawStr(0,40,String(WiFi.RSSI()).c_str());
  
  
  u8g2.sendBuffer();
  selectorMax = mesh.addrListTop;
  oldPos = newPos;
}
```

2bndy5, 2023-09-21:
try adding another `mesh.update()` before you write to your display:
```diff
+  mesh.update();
+  mesh.DHCP();
   u8g2.drawStr(0,10,WiFi.macAddress().c_str());
   u8g2.drawStr(0,20,String(selector).c_str());
   u8g2.drawStr(0,30,latestMessage.c_str());
   u8g2.drawStr(0,40,String(WiFi.RSSI()).c_str());
```
Did you first try the RF24Mesh (or RF24) library examples to make sure the radio works properly? This would be a good idea since it eliminates other code that might cause undesirable behavior.

---
BTW, if you're using Arduino IDE or VSCode, then it is a good habit to hit <kbd>Alt</kbd>+<kbd>Shift</kbd>+<kbd>F</kbd> after edits. This will run clang-format in Arduino IDE or whatever VSCode is setup to use as a C++ formatter. It will make your code much more readable.

ripjohnbrown1859, 2023-09-21:
im on platform io on vs code, i tried adding another ```mesh.update()```  right before my screen writes and it did not help.

2bndy5, 2023-09-21:
Is there only the 2 nodes in the mesh (including master)? I'm wondering if the transmission gets lost when hopping to intermediate nodes between master and your node using core 0.

Did you run the RF24Mesh example to be sure the radio works? If the example runs fine, then we know it isn't a RF24Mesh library problem and the Arduino Forums could be a better place for troubleshooting/questions about your situation.

As a last resort, you can enable the debugging logs in RF24Mesh_config.h:
https://github.com/nRF24/RF24Mesh/blob/38ee0225c336349fab513907f9798d308d2db497/RF24Mesh_config.h#L59-L60
And check the serial monitor to read the debugging output.

ripjohnbrown1859, 2023-09-21:
cant seem to get the library examples to work, any of them. heres the output from the mesh debug
ID: 1 ADDR: 05
Sent to 04444 phys: 04444 new: 05 id: 1

TMRh20, 2023-09-21:
You probably should troubleshoot using the core RF24 examples, then move on to RF24Mesh.

See https://github.com/nRF24/RF24/blob/master/COMMON_ISSUES.md for more info

ripjohnbrown1859, 2023-09-22:
i have 3 different types of nrf24 modules and i cant get 2 of the types(the standard pcb antenna kind and the unshielded +pa+lna) to even connect, the shielded units im using now are connecting but are not transmitting data.

2bndy5, 2023-09-22:
ESP32 boards typically consume a lot of power from the 3V regulator, and that leaves little power left for external hardware (like nRF24 radios). Try using an external power source for your nRF24 module when using the ESP32 as a driver.

> **Note**
> Some ESP32 boards from [Unexpected Maker](https://unexpectedmaker.com/) have a separate 3V regulator for external hardware because the ESP32 requires so much power.

ripjohnbrown1859, 2023-09-22:
I already am using a separate battery to power them.

2bndy5, 2023-09-22:
1. What exact nRF24 modules are we talking about? I assume the shielded PA/LNA modules are likely ebyte modules and the unshielded modules are those that are widely available from general purpose retailers (like Amazon). The non-PA/LNA modules should work well with other non-PA/LNA modules since their power requirements are much less.
2. Which library examples have you tried that are failing? Most of the examples use `RF24_PA_MIN` level to reduce the required power consumption (a soft hack best for PA/LNA modules).
3. This might seem silly to ask, but are the grounds tied together between different power sources?

ripjohnbrown1859, 2023-09-22:
the shielded ones are these
https://www.amazon.com/dp/B07QC1SXJ8

TMRh20, 2023-09-22:
There is a comment about resistors on the board for those sheilds in the reviews. Have you tried following it and adding `RF24 radio(7,8,1000000);` to your code?

You need to tie the grounds together.

2bndy5, 2023-09-22:
> the grounds are not tied together on the supplies.

They SHOULD be! This is basic understanding of electricity.

ripjohnbrown1859, 2023-09-22:
Oops everyone always specified having a separate power source for the nrf24. Totally didn't think about the grounds. Ill see of that fixes things

2bndy5, 2023-09-22:
[This reviewer points out why the resistors used on those boards require a lower SPI speed](https://www.amazon.com/gp/customer-reviews/RHF14KJILUSZW/ref=cm_cr_dp_d_rvw_ttl?ie=UTF8&ASIN=B07QC1SXJ8). The RF24 library default is 10 MHz, though reviewers recommend 1 MHZ. Thus the third parameter to the RF24 c'tor might need to be much less.

ripjohnbrown1859, 2023-09-22:
> [This reviewer points out why the resistors used on those boards require a lower SPI speed](https://www.amazon.com/gp/customer-reviews/RHF14KJILUSZW/ref=cm_cr_dp_d_rvw_ttl?ie=UTF8&ASIN=B07QC1SXJ8). [...]

I lowered the speed to 4mhz, i will try 1 mhz next

ripjohnbrown1859, 2023-09-22:
grounds tied together, spi speed set to 1 mhz, still no luck.

2bndy5, 2023-09-22:
How are you powering it? Specifically what voltage regulator? Have you added a capacitor(s) to the radio's VCC and GND lines?

ripjohnbrown1859, 2023-09-22:
Im using li ion vape batteries and a power breakout that uses an ams 1117 regulator.  I didnt see anything about needing a capacitor on the shielded modules but i can try adding one.

2bndy5, 2023-09-22:
That regulator should be supplying sufficient power, and it does require the breakout board to use capacitors. However, we still recommend an extra capacitor as close as possible to the radio (parallel to its VCC & GND) to ensure power stability. This recommendation is mentioned a few times in the RF24/COMMON_ISSUES.md doc.

Just to be sure, have you at any time powered the radio with more than 3.3v? This usually fries the radio, so please avoid that.

I assume have connected the antennas as well.

ripjohnbrown1859, 2023-09-22:
Ill add caps,  what capacitance do you recommend? 
Also ive only ever powered them through the regulator so never above 3.3v and i have connected the antennas.

2bndy5, 2023-09-22:
> Ill add caps, what capacitance do you recommend? 

https://github.com/nRF24/RF24/blob/master/COMMON_ISSUES.md#my-palna-module-doesnt-perform-as-well-as-id-hoped-or-the-nrf-radio-works-better-on-touching-it

ripjohnbrown1859, 2023-09-22:
added 10uf caps, still not sending data, gonna build a channel scanner to see whats going on.

2bndy5, 2023-09-22:
The RF24 library has an example called scanner. You could use that with you Serial Monitor to get an understanding of ambient noise. The Arduino scanner example also offers a way to instigate a carrier wave that should be seen by any radio scanning for ambient noise.

This carrier wave test is how manufacturers satisfy requirements mandated by the US FCC (or a different government agency in other countries).

You also might have to play with multiple different capacitance values since the value is subject to _your_ circuit's electrical properties.

ripjohnbrown1859, 2023-09-22:
surely it would at least work poorly if the capacitance was wrong.

2bndy5, 2023-09-22:
That's not guaranteed.

ripjohnbrown1859, 2023-09-22:
we are transmitting on the default channel it seems.

2bndy5, 2023-09-22:
This issue seems to be a typical power supply problems with PA/LNA modules.

> surely someone in the reviews would post something about how they dont work ...

Please don't rely on the goodwill of others, that would be naive. The fact that we're still helping you here is generous on our behalf because this has become a hardware issue, not a software issue. We (the nRF24 GitHub org) don't produce hardware, only software.

ripjohnbrown1859, 2023-09-22:
sorry, i very much appreciate your help. what should i do to remediate the power supply issue?

2bndy5, 2023-09-22:
Personally, I'd return the radio's using Amazon's refund policy, and buy from a different manufacturer. I think we have exhausted all advice that we usually tell people in this situation.

It's hard to find guaranteed working nRF24 hardware nowadays (since Nordic open sourced the PCB design in 2014). Ebyte sells a bunch of PA/LNA variants, but they are still subject to some quirks. Honestly, if you're not transmitting farther than 10-15 meters (less than 35 ft), then you would typically have a better experience with nRF24 modules that have built-in antennas. The PA/LNA modules are a good example of when "bigger" is not synonymous with "better".

ripjohnbrown1859, 2023-09-22:
i need need much longer range than that, ideally up to a kilometer. the thing im stuck on is that an address gets assigned to the node, and theres actual transmission happening. throughout this process ive now gotten all the different chips to send data, its just not being recieved

2bndy5, 2023-09-22:
The mesh handshake process does not use the radio's auto-ack, but all other network/mesh transmissions do.

Usually, failed auto-acking from a radio means it did not have enough current to transmit. Sometimes, Chinese clones (like the infamously bad Si24R1) are physically flawed to the point where the auto-ack feature doesn't work at all.

We don't recommend disabling the auto-ack because it _will_ lead to packet loss. RF24 network layers fundamentally use auto-ack for reliability/stability.

ripjohnbrown1859, 2023-09-22:
i disabled auto ack to see if that was it but they still are not receiving data

2bndy5, 2023-09-22:
it has to disabled on both ends

ripjohnbrown1859, 2023-09-22:
it is disabled on both ends

2bndy5, 2023-09-22:
Then there's something seriously wrong with those modules. Have tried on different channels?

I'm stepping out now, so I'll be radio silent for a while...

ripjohnbrown1859, 2023-09-22:
ok. ill try on dif channels, thx for the help.

TMRh20, 2023-09-23:
Have you enabled printf per https://github.com/nRF24/RF24/blob/master/COMMON_ISSUES.md and verified you have the radios hooked up correctly?

ripjohnbrown1859, 2023-09-23:
SPI Speedz      = 1 Mhz
STATUS          = 0x0e RX_DR=0 TX_DS=0 MAX_RT=0 RX_P_NO=7 TX_FULL=0
RX_ADDR_P0-1    = 0xcccccc3ccc 0xcccccce33c
RX_ADDR_P2-5    = 0x33 0xce 0x3e 0xe3
TX_ADDR         = 0xcccccccce3
RX_PW_P0-6      = 0x20 0x20 0x20 0x20 0x20 0x20
EN_AA           = 0x3f
EN_RXADDR       = 0x03
RF_CH           = 0x4c
RF_SETUP        = 0x07
CONFIG          = 0x0e
DYNPD/FEATURE   = 0x00 0x00
Data Rate       = 1 MBPS
Model           = nRF24L01+
CRC Length      = 16 bits
PA Power        = PA_MAX
ARC             = 0

ripjohnbrown1859, 2023-09-23:
somethings up might be due to the alternate spi bus?

ripjohnbrown1859, 2023-09-23:
ditto reciever which is using the main spi bus
SPI Speedz	= 1 Mhz
STATUS		= 0x0e RX_DR=0 TX_DS=0 MAX_RT=0 RX_P_NO=7 TX_FULL=0
RX_ADDR_P0-1	= 0xccccccccc3 0xcccccccc3c
RX_ADDR_P2-5	= 0x33 0xce 0x3e 0xe3
TX_ADDR		= 0xe7e7e7e7e7
RX_PW_P0-6	= 0x20 0x20 0x20 0x20 0x20 0x20
EN_AA		= 0x3e
EN_RXADDR	= 0x3f
RF_CH		= 0x61
RF_SETUP	= 0x01
CONFIG		= 0x0f
DYNPD/FEATURE	= 0x3f 0x04
Data Rate	= 1 MBPS
Model		= nRF24L01+
CRC Length	= 16 bits
PA Power	= PA_MIN
ARC		= 0

TMRh20, 2023-09-23:
I would guess you have bad radios, but multiple different kinds? That seems odd. It could still be your CS/CE pins are wrong, but again I doubt it. I'm kind of stumped on this one.

ripjohnbrown1859, 2023-09-23:
3 different kinds and the scanner demo works! idk what is going on here.  I've noticed that the node connects only when the master is a shielded module, but regardless of the kind of module the node is. also when i disable autoAck it does not connect. either way whether or not it connects no data seems to be recieved.

ripjohnbrown1859, 2023-09-23:
the scanner demo works on any of the modules, without a capacitor, powered directly of the board.

2bndy5, 2023-09-23:
I googled ML01DP5 and found the following spec sheet [from ebyte](https://www.cdebyte.com/products/E01-ML01DP5):
[E01-ML01DP5_Usermanual_EN_V1.3.pdf](https://github.com/nRF24/RF24Mesh/files/12705396/E01-ML01DP5_Usermanual_EN_V1.3.pdf)

Pay close attention to anything that mentions the power supply. For instance:
[image]

---
MakerFocus may be redistributor of the ebyte modules or a ebyte clone. Its hard to say with Chinese hardware at a glance; the pictures on the Amazon store page are not consistent and probably a poor representation of the actual product.

2bndy5, 2023-09-23:
I'm pretty sure the ebyte modules use a fixed PA level, so using `radio.setPALevel()` probably has no affect on these modules. Notice the table near the bottom of the spec sheet:
[image]
I believe this means that it only uses 20dbm PA level.

ripjohnbrown1859, 2023-09-23:
but even when i use the pcb antenna on both ends, nothing works.

ripjohnbrown1859, 2023-09-23:
i just got a demo to work! the getting started demo in rf24 lib works now.  still cant get the mesh demos to work though.

2bndy5, 2023-09-23:
🎉 sounds like your close to expected behavior. I'm curious if you know what you did differently to get it to work.

> mesh demo started working intermittently

Can you describe this more?

ripjohnbrown1859, 2023-09-23:
i changed it to channel 100, also one of my cats knocked it off the table at some point and the solder joint on the capacitor broke without me noticing so i fixed that. 
the mesh is only getting maybe 50% of the packets through, and it is still not working on my program.

2bndy5, 2023-09-23:
Ah yes, cats + DIY electronics = rework; I know this too well.

Unfortunately, I don't have any fresh advice.

ripjohnbrown1859, 2023-09-23:
I suspect it could be the voltage drop across the regulator from 4.2 volts to 3.1v

2bndy5, 2023-09-23:
Its funny you mention that. Another user that bought modules from ebyte directly found that his battery would cause brown outs after a few hours as the battery discharged. While these brown outs, went unnoticed, he had a bad habit of calling `radio.begin()` to keep the drone going... I think he doubled up on the battery pack (2 in parallel) to ensure the battery discharge remained above optimal levels.

Not sure if any of that would be helpful to you.

ripjohnbrown1859, 2023-09-24:
ok so ive gotten the demo to work with the ebyte modules 100%.  but still no luck on my code.

2bndy5, 2023-09-24:
That may be a task priority problem with the RTOS. I can't help you at all with that.

ripjohnbrown1859, 2023-09-24:
its the only thing running on core 0, theres no wifi running, so i dont think its that.

2bndy5, 2023-09-24:
Well, now that we know that the radio is working, I would suggest seeking advice about using the ESP32 dual core/multiprocessing from people that are more knowledgeable on that subject.

Lastly, RF24Network/RF24Mesh nodes should not constantly spam messages out as that would clog the network/mesh channel which essentially blocks other nodes from being able to transmit/respond. Instead, users should regulate their nodes' outgoings with timed intervals (as is done on the examples).

ripjohnbrown1859, 2023-09-24:
what is the biggest size struct you can send? i wonder if im trying to send too much data at once?

2bndy5, 2023-09-24:
Payloads over 24 bytes get split up into multiple "frames" (separate transmissions) by the RF24Network layer.
Default max size is 144 bytes on non-linux platforms (1514 for Linux platforms). The required `RF24NetworkHeader` occupies 8 bytes.

[24 more replies]

closed 2023-09-27 by ripjohnbrown1859