nRF24/RF24Mesh #228: nrf_to_nrf dependency is causing compiler errors in platformio

shyney7, 2023-11-14:
The nrf_to_nrf library dependency is causing compiler errors like
`nrf_to_nrf/src/nrf_to_nrf.cpp:102:5: error: 'NRF_RADIO' was not declared in this scope`
to occur. This library is installed automatically when installing the RF24Mesh library in platformio.
I've tried several versions of the nrf_to_nrf library but the only solution was to delete the library completely.

<details>
  <summary>console output</summary>
  
  ### compiler errors:
  ```shell
   *  Executing task: C:\Users\olivei_m\.platformio\penv\Scripts\platformio.exe run 

Processing lilygo-t-display-s3 (platform: espressif32; board: lilygo-t-display-s3; framework: arduino)
-----------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------Verbose mode can be enabled via `-v, --verbose` option
CONFIGURATION: https://docs.platformio.org/page/boards/espressif32/lilygo-t-display-s3.html
PLATFORM: Espressif 32 (6.2.0) > LilyGo T-Display-S3
HARDWARE: ESP32S3 240MHz, 320KB RAM, 16MB Flash
DEBUG: Current (cmsis-dap) External (cmsis-dap, esp-bridge, esp-builtin, esp-prog, iot-bus-jtag, jlink, minimodule, olimex-arm-usb-ocd, olimex-arm-usb-ocd-h, olimex-arm-usb-tiny-h, olimex-jtag-tiny, tumpa)
PACKAGES:
 - framework-arduinoespressif32 @ 3.20008.0 (2.0.8)
 - tool-esptoolpy @ 1.40501.0 (4.5.1)
 - toolchain-riscv32-esp @ 8.4.0+2021r2-patch5
 - toolchain-xtensa-esp32s3 @ 8.4.0+2021r2-patch5
LDF: Library Dependency Finder -> https://bit.ly/configure-pio-ldf
LDF Modes: Finder ~ chain, Compatibility ~ soft
Found 41 compatible libraries
Scanning dependencies...
Dependency Graph
|-- CircularBuffer @ 1.3.3
|-- Button2 @ 2.2.4
|-- RF24Mesh @ 2.0.0
|-- nrf_to_nrf @ 1.2.6
|-- TFT_eSPI @ 2.5.0
|-- TouchLib @ 0.0.1
|-- RF24 @ 1.4.8
|-- RF24Network @ 2.0.0
|-- SPI @ 2.0.0
|-- Wire @ 2.0.0
Building in release mode
Compiling .pio\build\lilygo-t-display-s3\src\main.cpp.o
Building .pio\build\lilygo-t-display-s3\bootloader.bin
Generating partitions .pio\build\lilygo-t-display-s3\partitions.bin
esptool.py v4.5.1
Creating esp32s3 image...
Merged 1 ELF section
Successfully created esp32s3 image.
Compiling .pio\build\lilygo-t-display-s3\libfdd\Button2\Button2.cpp.o
Compiling .pio\build\lilygo-t-display-s3\libe64\SPI\SPI.cpp.o
Compiling .pio\build\lilygo-t-display-s3\lib69b\RF24\RF24.cpp.o
Compiling .pio\build\lilygo-t-display-s3\lib6ef\nrf_to_nrf\nrf_to_nrf.cpp.o
In file included from lib/TFT_eSPI/TFT_eSPI.h:94,
                 from include/graph.h:16,
                 from src/main.cpp:9:
lib/TFT_eSPI/Processors/TFT_eSPI_ESP32_S3.h:11:3: warning: #warning >>>>------>> DMA is not supported on the ESP32 S3 (possible future update) [-Wcpp]
  #warning >>>>------>> DMA is not supported on the ESP32 S3 (possible future update)
   ^~~~~~~
In file included from lib/TFT_eSPI/TFT_eSPI.h:94,
                 from include/graph.h:16,
                 from src/main.cpp:9:
lib/TFT_eSPI/Processors/TFT_eSPI_ESP32_S3.h:369:52: warning: backslash and newline separated by space
     xset_mask[c] = 0;                              \

lib/TFT_eSPI/Processors/TFT_eSPI_ESP32_S3.h:377:66: warning: backslash and newline separated by space
     if ( c & 0x80 ) xset_mask[c] |= (1 << (TFT_D7-MASK_OFFSET)); \

In file included from include/graph.h:16,
                 from src/main.cpp:9:
lib/TFT_eSPI/TFT_eSPI.h:953:8: warning: #warning >>>>------>> TOUCH_CS pin not defined, TFT_eSPI touch functions will not be available! [-Wcpp]
       #warning >>>>------>> TOUCH_CS pin not defined, TFT_eSPI touch functions will not be available!
        ^~~~~~~
[... 672 lines cut ...]
                ^~
lib/TFT_eSPI/Processors/TFT_eSPI_ESP32_S3.c:110:16: note: in expansion of macro 'TFT_D4'
   b |= (((reg>>TFT_D4)&1) << 4);
                ^~~~~~
lib/TFT_eSPI/User_Setups/Setup206_LilyGo_T_Display_S3.h:31:16: warning: right shift count >= width of type [-Wshift-count-overflow]
 #define TFT_D5 46
                ^~
lib/TFT_eSPI/Processors/TFT_eSPI_ESP32_S3.c:111:16: note: in expansion of macro 'TFT_D5'
   b |= (((reg>>TFT_D5)&1) << 5);
                ^~~~~~
lib/TFT_eSPI/User_Setups/Setup206_LilyGo_T_Display_S3.h:32:16: warning: right shift count >= width of type [-Wshift-count-overflow]
 #define TFT_D6 47
                ^~
lib/TFT_eSPI/Processors/TFT_eSPI_ESP32_S3.c:112:16: note: in expansion of macro 'TFT_D6'
   b |= (((reg>>TFT_D6)&1) << 6);
                ^~~~~~
lib/TFT_eSPI/User_Setups/Setup206_LilyGo_T_Display_S3.h:33:16: warning: right shift count >= width of type [-Wshift-count-overflow]
 #define TFT_D7 48
                ^~
lib/TFT_eSPI/Processors/TFT_eSPI_ESP32_S3.c:113:16: note: in expansion of macro 'TFT_D7'
   b |= (((reg>>TFT_D7)&1) << 7);
                ^~~~~~
============================================================================== [FAILED] Took 18.80 seconds ==============================================================================
  ```
</details>

2bndy5, 2023-11-14:
RF24Mesh doesn't actually require nrf_to_nrf lib installed (as you can check in [RF24Mesh's v2.0.0 library.json](https://github.com/nRF24/RF24Mesh/blob/master/library.json). However, [RF24Network's v2.0 library.json](https://github.com/nRF24/RF24Network/blob/master/library.json) does name a dependency on nrf_to_nrf library for the `nordicnrf52` platform. This shouldn't be applicable for apps built on other platforms (like ESP32 in your case), and the `#include <nrf_to_nrf.h>` is guarded with
https://github.com/nRF24/RF24Mesh/blob/38ee0225c336349fab513907f9798d308d2db497/RF24Mesh.h#L44-L46

For the [nrf_to_nrf v1.2.2 library.json](https://github.com/TMRh20/nrf_to_nrf/blob/v1.2.2/library.json), there was a named dependency on a certain branch of RF24Mesh, but that dependency specification has been removed as of [v1.2.6 library.json](https://github.com/TMRh20/nrf_to_nrf/blob/v1.2.6/library.json) (& the RF24Mesh branch has been deleted).

1. Can you provide your project's platformio.ini file? This case was a specific concern when we released v2.0 of the network layers.
2. Are you installing libraries globally in PIO? The build logs you posted don't seem to indicate this, but I thought I'd ask anyway.
3. Lastly, what version of PlatformIO are you using? I'm wondering if an older version does not respect the `platform` field of the nrf_to_nrf dependency in RF24Network's v2.0 library.json.

shyney7, 2023-11-14:
1. 
```bash
[env:lilygo-t-display-s3]
platform = espressif32
board = lilygo-t-display-s3
framework = arduino
lib_deps = 
	rlogiacco/CircularBuffer@^1.3.3
	lennarthennigs/Button2@^2.2.4
	nrf24/RF24Mesh@^2.0.0
build_flags = '-DARDUINO_USB_MODE=1' '-DARDUINO_USB_CDC_ON_BOOT=1'
```
2. all my libraries are installed project wise
3. Core 6.1.11·Home 3.4.4

2bndy5, 2023-11-14:
Thanks for answering my questions. I don't see any problems other than the platform-specific dependency getting installed (which is not intended).

I feel like this is a bug in PIO core itself. I see they had [a similar problem in the past](https://github.com/platformio/platformio-core/issues/4338), but that was fixed in v6.1.0.

According to [the docs about library dependencies](https://docs.platformio.org/en/latest/manifests/library-json/fields/dependencies.html), we are doing everything right as far as I can tell.

2bndy5, 2023-11-14:
Does it help to add RF24 & RF24Network to your project's `lib_deps` (preferably before RF24Mesh in the list)?
```ini
lib_deps = 
  rlogiacco/CircularBuffer@^1.3.3
  lennarthennigs/Button2@^2.2.4
  nRF24/RF24@^1.4.8
  nRF24/RF24Network@^2.0.0
  nrf24/RF24Mesh@^2.0.0
```

Maybe the PIO LDF is losing the platform info as it seeks dependencies of dependencies.

shyney7, 2023-11-16:
@2bndy5 
Ok I did a full clean of the build environment and everything works now. I really dont know what caused this behavior in the first place!? Thank you for your support.

closed 2023-11-16 by shyney7

2bndy5, 2023-11-16:
Still its good that reported it. At least I have better advice now if someone else comes across this problem.

NorbertSandor, 2023-12-17:
> https://github.com/nRF24/RF24Mesh/issues/228#issuecomment-1811431278

This workaround does not work for me :(

> RF24Mesh doesn't actually require nrf_to_nrf lib installed

Based on this comment, I tried

```
lib_ignore =
	nrf_to_nrf
```

and the code compiles.
Do you think it is correct to exclude `nrf_to_nrf` altogether?

2bndy5, 2023-12-17:
It is certainly an acceptable workaround for projects that don't use nRF52 boards, but we made sure the dependencies are specifically and conditionally stated. 

@NorbertSandor I have some questions:
1. What is your platformio.ini file look like?
2. Are you sure you completely removed/deleted your .pio folder in the project root folder?
3. What version of PlatformIO core are you using?

NorbertSandor, 2023-12-17:
@2bndy5 Deleting the `.pio` folder solved the problem, sorry for being inattentive :|
Now it compiles without `lib_ignore` - thanks.