nRF24/RF24Mesh #230: Can a master Tx to a slave in an nRF24 mesh?

Kim-SangYoul, 2024-01-12:
Hi. I want to implement 1:n communication using nRF24 Mesh, but the current example code only allows n slaves to transmit to 1 master. I would like to add to this a TX transmission from the master back to a specific slave. Is this possible? If possible, a reference code would be appreciated. Thank you.

2bndy5, 2024-01-12:
like in this example? [RF24Mesh_Example_Master_To_Nodes.ino](https://github.com/nRF24/RF24Mesh/blob/master/examples/RF24Mesh_Example_Master_To_Nodes/RF24Mesh_Example_Master_To_Nodes.ino)

Kim-SangYoul, 2024-01-12:
Thank you for your quick response. RF24Mesh_Example_Master_To_Nodes.ino example name looks like it is sending from master to nodes, but how can I send from master to TX?

2bndy5, 2024-01-12:
>how can I send from master to TX?

I'm not sure you are wording this question the way you want.

The nRF24L is a transceiver, so every node is a TX and a RX. With the network layers, the nodes idle in RX so they can respond (TX) in a timely manner.

More [basic topology info is available in the python binding's docs](https://nrf24.github.io/pyRF24/topology.html).

Kim-SangYoul, 2024-01-12:
Thank you. I am a hardware engineer so my understanding is limited. How can I implement the ? part in the attached picture if it is possible. Please explain in a little more detail. Thank you.
[image]

2bndy5, 2024-01-12:
No need to send to master and then have master send to another node. You can just send directly to the destination.

[Using RF24Network address](https://nrf24.github.io/RF24Mesh/classESBMesh.html#a39eec0fb9900722af8646a5a88df90e7)
```c++
// from node 05314
char msgType = 'T';
mesh.write(0421, &buf, msg_type, sizeof(buf));
```
[Using RF24Mesh ID](https://nrf24.github.io/RF24Mesh/classESBMesh.html#aceefd01852997514bdb5bc8f89e7ca55)
```c++
// from 05314
// assuming 0421 is assign to node with ID 50
mesh.write(&buf, msg_type, sizeof(buf), 50);
```

All nodes will automatically do the forwarding for you. This is explained in the doc I linked:

> For a message to travel from node `0o124` to node `0o3`, it must be passed through any applicable network levels. So, the message flows `0o124` -> `0o24` -> `0o4` -> `0o0` -> `0o3`.

Maybe the wording is confusing there.

Kim-SangYoul, 2024-01-12:
Thank you. I didn't know the basics. A simple motion has been confirmed.

closed 2024-01-12 by 2bndy5